Opentrons/opentrons #12318: bug: API bug?

rickwierenga, 2023-03-20:
### Overview

I'm trying to aspirate using [the HTTP API](https://github.com/Opentrons/opentrons/tree/edge/api-client), but get an error `UnexpectedProtocolError: Cannot aspirate more than pipette max volume`. A p20 pipette is successfully loaded, and the aspiration is 5.0 uL, so this error is unexpected.

### Steps to reproduce

The issue is described in more detail in https://github.com/PyLabRobot/pylabrobot/issues/23.

Specifically, see the following mitmproxy dump: [data_20_03_2023.zip](https://github.com/Opentrons/opentrons/files/11019309/data_20_03_2023.zip)

### Operating system

None

SyntaxColoring, 2023-03-20:
Thanks for reporting this. This seems strange. We'll look into this and update here when we have more information.

rickwierenga, 2023-03-21:
Turn out to be an issue in an external library. Thanks again for the help!

closed 2023-03-21 by rickwierenga