Opentrons/opentrons #12397: feat: unload labware

arogozhnikov, 2023-03-31:
### Overview

I want to run a protocol in stages. Example of such stages:

Stage 1: preparation of dilutions
Stage 2: distribute into plates A, B
Stage 3: distribute into plates C, D

Plates C and D go to the same slots on deck as A and B.

Advantage of having stages, not several independent protocols: contents of some intermediate plates (dilutions plate in the example above) is kept in program state, and can be done once.

I think a simple way would be to allow 'unload labware' or 'replace labware', which will also invalidate any further operations with labware object that was unloaded. 
Alternatively, a flag  `replace_existing_labware=True` for `load_labware`  may signify that we expect some labware being already on this spot and that it should be unloaded at this moment. 

### Acceptance criteria

Being able to swap labware in the middle of the run.

jeffhsu3, 2023-04-17:
Would love this as well.

SyntaxColoring, 2024-02-29:
👍 See https://github.com/Opentrons/opentrons/issues/6214#issuecomment-1957857284.

closed 2024-02-29 by SyntaxColoring