Opentrons/opentrons #12476: bug:

busing, 2023-04-12:
### Overview

When I ran “Run Labware Position Check” with source code, the interface did not respond and the console reported an error

### Steps to reproduce

[image]

### Operating system

None

koji, 2023-04-15:
Hi @busing,

Is it possible to share your protocol file with us?

busing, 2023-04-17:
Thans for your reply.
Sure , here is my protocol file ,  Actually, I won't encounter this issue by downloading the installation file directly, but this error will occur when running with the source code，Did I miss any steps when running the code?
`
from opentrons import protocol_api
import json
import urllib.request
import urllib.parse
import urllib.error
import socket

metadata = {
    'protocolName': '2023-03-14-test',
    'author': '邰元(ytai@terrydr.com)',
    'description': '2023-03-14-test',
    'apiLevel': '2.7'

}
####################MODIFIABLE RUN PARAMETERS#########################
default_flow_rate = 50
tiprack_starting_pos = {
    "tiprack_10": 'A1',
    "tiprack_300": 'A1'
}

corning_96_wellplate_360ul_flat = 'corning_96_wellplate_360ul_flat'
type_corning_6_wellplate_168ml_flat = 'corning_6_wellplate_16.8ml_flat'
corning_96_list = ['A1']

loopTimes = 1

# debug mode skips all incubations, prints out additional info
debug = True

####################FIXED RUN PARAMETERS#########################

API_VERSION = '2.7'

class Object:
    pass

####################LABWARE LAYOUT ON DECK#########################
labwarePositions = Object()
labwarePositions.corning_6_wellplate_168ml_flat = 1
labwarePositions.corning_96 = 2
labwarePositions.tiprack_300 = 6
####################GENERAL SETUP################################

def run(protocol: protocol_api.ProtocolContext):

    if debug: print(protocol)
    # |---|
    tiprack_300 = protocol.load_labware('opentrons_96_tiprack_300ul', labwarePositions.tiprack_300, "tiprack 300ul")
    corning_6_wellplate_168ml_flat = protocol.load_labware(type_corning_6_wellplate_168ml_flat, labwarePositions.corning_6_wellplate_168ml_flat, type_corning_6_wellplate_168ml_flat)
    corning_96 = protocol.load_labware(corning_96_wellplate_360ul_flat, labwarePositions.corning_96, corning_96_wellplate_360ul_flat)
    if debug: print(tiprack_300)

    # 移液器
    pipette_300 = protocol.load_instrument('p300_single', 'left', tip_racks=[tiprack_300])
    pipette_300.flow_rate.dispense = default_flow_rate
    pipette_300.flow_rate.aspirate = default_flow_rate
    # 开始位置
    pipette_300.starting_tip = tiprack_300.well(tiprack_starting_pos['tiprack_300'])

    if debug: print(pipette_300)

    # 吸管
    pipette_300.pick_up_tip()

    for i in range(loopTimes):
        for hole in corning_96_list:
            print (hole)
            pipette_300.aspirate(160, corning_6_wellplate_168ml_flat['A1'])
            pipette_300.dispense(160, corning_96[hole])
    pipette_300.drop_tip()
`

koji, 2023-04-20:
Hi @busing 
Sorry for my slow response.
I tried to reproduce the issue you faced on my end, but I couldn't reproduced that.
Could you check 2 thing for me?
1. Is the following code the same as you tried? (I took this from your post)
2. Could you tell me your robot's software version and desktop version.

Robot software version: Click a robot in Devices page -> click 3 dots menu -> select `Robot settings` -> click `Advanced` tab
Desktop version: Click the gear icon in the sidebar 

Thank you.

```py
from opentrons import protocol_api
import json
import urllib.request
import urllib.parse
import urllib.error
import socket

metadata = {
'protocolName': '2023-03-14-test',
'author': 'xxxxx',
'description': '2023-03-14-test',
'apiLevel': '2.7'

}
####################MODIFIABLE RUN PARAMETERS#########################
default_flow_rate = 50
tiprack_starting_pos = {
"tiprack_10": 'A1',
"tiprack_300": 'A1'
}

corning_96_wellplate_360ul_flat = 'corning_96_wellplate_360ul_flat'
type_corning_6_wellplate_168ml_flat = 'corning_6_wellplate_16.8ml_flat'
corning_96_list = ['A1']

loopTimes = 1

debug = True

####################FIXED RUN PARAMETERS#########################

API_VERSION = '2.7'

class Object:
    pass

####################LABWARE LAYOUT ON DECK#########################
labwarePositions = Object()
labwarePositions.corning_6_wellplate_168ml_flat = 1
labwarePositions.corning_96 = 2
labwarePositions.tiprack_300 = 6
####################GENERAL SETUP################################

def run(protocol: protocol_api.ProtocolContext):

    if debug: print(protocol)
    # |---|
    tiprack_300 = protocol.load_labware('opentrons_96_tiprack_300ul', labwarePositions.tiprack_300, "tiprack 300ul")
    corning_6_wellplate_168ml_flat = protocol.load_labware(type_corning_6_wellplate_168ml_flat, labwarePositions.corning_6_wellplate_168ml_flat, type_corning_6_wellplate_168ml_flat)
    corning_96 = protocol.load_labware(corning_96_wellplate_360ul_flat, labwarePositions.corning_96, corning_96_wellplate_360ul_flat)
    if debug: print(tiprack_300)

    # 移液器
    pipette_300 = protocol.load_instrument('p300_single', 'left', tip_racks=[tiprack_300])
    pipette_300.flow_rate.dispense = default_flow_rate
    pipette_300.flow_rate.aspirate = default_flow_rate
    # 开始位置
    pipette_300.starting_tip = tiprack_300.well(tiprack_starting_pos['tiprack_300'])

    if debug: print(pipette_300)

    # 吸管
    pipette_300.pick_up_tip()

    for i in range(loopTimes):
        for hole in corning_96_list:
            print (hole)
            pipette_300.aspirate(160, corning_6_wellplate_168ml_flat['A1'])
            pipette_300.dispense(160, corning_96[hole])
    pipette_300.drop_tip()
```

busing, 2023-04-21:
Thanks for your reply .

- My robot version is v6.2.1
- Desktop version is v6.2.1 too
- Protocol is same as me
- Whether I am running code from the Edga branch or v6.2.1 or v6.3.1tag, I still have this issue
- There is no problem running the protocol, as an error was reported when clicking on "Run Labware Position Check". No matter what protocol it is, clicking on "Run Labware Position Check" will not work
- Running the app using Github's source code encountered this issue, which was not found after installing the exe downloaded from the official website

[image]

koji, 2023-04-25:
If you need this, please re-open!

closed 2023-04-25 by koji