Opentrons/opentrons #12520: bug: air_gap causes 'Cannot aspirate more than pipette max volume'

FabianRoger, 2023-04-18:
### Overview

When doing a transfer of a larger volume broken down into multiple steps, adding an air_gap causes the 'Cannot aspirate more than pipette max volume' error to be raised

### Steps to reproduce

I tried to reduce it to a minimal example, see attached file. Removing ```pipette_left.air_gap(10)```on line 66 causes teh code to simulate without problem, keeping it raises the error - independent of the value set for tip_vol. 

[mag_bead_DEBUG.txt](https://github.com/Opentrons/opentrons/files/11266689/mag_bead_DEBUG.txt)

### Current behavior

raises an "Cannot aspirate more than pipette max volume" when calling an air_gap even though the volumes + airgap are below the max volume. 

### Expected behavior

it shouldn't raise an error

### Operating system

Mac

### System and robot setup or anything else?

error occurs in when simulating (on a mac) as well as on the robot (windows + usb connection)

koji, 2023-04-21:
@FabianRoger 
Could you reach out to our support team [support@opentrons.com](mailto:support@opentrons.com), please?

koji, 2023-04-23:
If needed, please re-open.

closed 2023-04-23 by koji

FabianRoger, 2023-08-09:
FYI: 

The problem is caused by the second air_gap because the plunger is apparently not reset before the new transfer. this means that if a large transfer is broken into multiple steps, the air gap accumulates until the pipette runs out of volume.