Opentrons/opentrons #12560: bug: can't set defaults in well_bottom_clearance

arogozhnikov, 2023-04-25:
### Overview

Exact bug is this (quoting the documentation):

```python
# To change, set the corresponding attribute. For instance,
instr.well_bottom_clearance.aspirate = 1
```

This will not work:
```
pipette.well_bottom_clearance.aspirate = 10
print(pipette.well_bottom_clearance.aspirate) # will print 1.
```

Reason is hidden in the implementation: 
```
pipette.well_bottom_clearance -> calls self._implementation.get_well_bottom_clearance()
```
which has following implementation:

```python
def get_well_bottom_clearance(self) -> Clearances:
    return Clearances(default_aspirate=1, default_dispense=1)
```
i.e. each time it creates a new object, and changing returned object has no effect.

A simple solution could be to add `lru_cache` decorator on `get_well_bottom_clearance`.

### Steps to reproduce

See above

### Current behavior

See above

### Expected behavior

See above

### Operating system

Mac

### System and robot setup or anything else?

Brew-installed python 3.10
opentrons==6.2.1

arogozhnikov, 2023-04-25:
My team wants to use `pipette.distribute` with custom dispense/aspirate height, and changing defaults for pipette seem to be the only possible way.

arogozhnikov, 2023-04-25:
Update: it looks like location can be passed to distribute, but typing says that only wells are accepted:

https://github.com/Opentrons/opentrons/blob/e32ef94a8c64a890fbeb0f1aabfe71464390db77/api/src/opentrons/protocol_api/instrument_context.py#L972

ecormany, 2024-08-09:
Not sure when it happened, but we appear to have fixed this.

closed 2024-08-09 by arogozhnikov