Opentrons/opentrons #12683: bug: labware collision during Labware Position Check in v6.3.0

agostinofacotti, 2023-05-11:
### Overview

When running Labware Position Check the pipette will collide with labware higher than the exiting height from the trash.
Seen with OT and Opentrons App v6.3.0

Seems related to #12586

### Steps to reproduce

Open the OT app, load the attached protocol and than execute the Labware Position Check
[collision_during_calibration_with_6-3.zip](https://github.com/Opentrons/opentrons/files/11452771/collision_during_calibration_with_6-3.zip)

### Current behavior

When calibrating the NEST plate in slot 4 the P300 pipette with tip will collide with the 1000ul tiprack in slot 8.
Similar behaviour when going back from slot 4 to trash after having confirmed the calibration.

### Expected behavior

The pipette should rise enough not to collide with labware.

### Operating system

Windows

### System and robot setup or anything else?

App v6.3.0

randolba, 2023-06-15:
Hello, were you able to resolve this issue? We are experiencing the same issue here when I go to teach labware positions, the temperature module is in position 10 and the tip box is in position 7 and when it goes to it, it will crash into the temperature module.

agostinofacotti, 2023-06-19:
Seems that no one is taking care about that on OT2 for more than a month.
The labware calibration in 6.3.0 IMO is unacceptable, I'll downgrade the robot to v6.2.1 asap

randolba, 2023-06-19:
Hi I was able to reach out and contact Opentrons on a Pylab Forum. Here is the link with a solution. https://forums.pylabrobot.org/t/opentrons-ot-2-issue/1697