Opentrons/opentrons #12777: bug: I cannot make setup in the opentrons-edge folder

YiqiaoSun, 2023-05-24:
### Overview

Hi, 
I am new to labware creator. 
When I run make setup in the opentrons-edge folder I have the following error: 

opentrons-edge % make setup
yarn config set network-timeout 60000
make: yarn: No such file or directory
make: *** [setup-js] Error 1

How should I do about it? 

### Operating system

None

koji, 2023-05-24:
@YiqiaoSun 
Did you follow the [setup guide](https://github.com/Opentrons/opentrons/blob/edge/DEV_SETUP.md)?

If not, please try the link.
If you followed the guide and got the error, would you tell me your environment info?

- your os and os version
- pyenv version
- python version
- nodejs version currently we only support v16
- yarn version our repo supports v1 only

YiqiaoSun, 2023-05-25:
Hi Koji, 

Thank you very much for the reply. Now I have followed the setup guide and I got things right.

closed 2023-05-25 by YiqiaoSun