Opentrons/opentrons #12910: bug: scp not working on opentrons2

arogozhnikov, 2023-06-13:
### Overview

Scp used to work previously, but seems recently was broken.

### Steps to reproduce

```
ssh ot2 # logs in successfully
<closing ssh session>
```

however scp does not work:
```
scp some-local-file ot2:some-server-location
```
sh: /usr/libexec/sftp-server: not found
scp: Connection closed

### Operating system

Mac

### System and robot setup or anything else?

app and ot2 formware: 6.3.1
mac's ssh works ok (checked transfers on a different remote host)

koji, 2023-06-14:
have you tried to add `-o` option to use your ssh option?

arogozhnikov, 2023-06-14:
No `-o` options were used, all details are in ssh config.

Here is the reason: 
https://bugs.archlinux.org/task/74649

New ssh (9.0 or newer, default on mac) does not use legacy sftp, and requires a standalone sftp-server (which OT2 misses).

Currently a solution is to forcefully switch to legacy protocol with `-O`:
```
scp -O ot2:<path1>  <local_path>
```

There is [no way](https://lists.mindrot.org/pipermail/openssh-unix-dev/2022-June/040320.html) to put this information in ssh config.

SyntaxColoring, 2024-02-20:
I think this will be fixed in v7.2.0, as part of general system updates.