Opentrons/opentrons #12952: bug: labware error when trying to "proceed to run" prevents run

cemull3, 2023-06-22:
### Overview

Whenever I try to run a protocol, I receive this error, even when the protocol does not use this labware and this labware has been removed from the library: 
opentrons.protocol_reader.file_identifier.FileIdentificationError: thermoscientific_glasscoated_u.shaped_300ul_96well.json is not valid JSON. Expecting value: line 1 column 1 (char 0)

### Steps to reproduce

Upload and Attempt to run any protocol and hit "proceed to set up"

Prior to this, the Opentron had a protocol that was cancelled prematurely. The robot and app has been restarted multiple times. The computer also has been restarted. 

### Current behavior

Unable to run any protocol
Robot does not seem to have reset itself after the last canceled protocol

### Expected behavior

Robot should reset itself, with pipets moving back into position over waste
Protocol should run, unobstructed by a labware definition that is not used in the protocol or even in the library.

### Operating system

Windows

### System and robot setup or anything else?

USB connection
Version 6.3.1

KU-JonasL, 2023-07-03:
I have a similar experience. I had a protocol loading on the OT-2, but has the OT-2 was not responsive I cancelled the protocol. Unfortunately, I now get a similar error, though my "file_identifier" complains about another file (custom labware for LVL tubes). 

I have tried to remove the custom json and add it back to the library (testing in between). I have restarted the PC, the OT-2 and reinstalled the OT-2 software without luck. The same protocols I want to up on it, can be analysed and opened on our other OT-2

cemull3, 2023-07-03:
Hi!

I was finally able to solve the problem by performing a factory reset:

Factory resetting the OT-2 (opentrons.com)<https://support.opentrons.com/s/article/Factory-resetting-the-OT-2>

Hope this helps!

Catherine Mullins

Research Specialist
Gangarosa Department of Environmental Health
Rollins School of Public Health, Emory University
Atlanta, GA

KU-JonasL, 2023-07-04:
Thank you Catherine! It helped me as well, I can now load protocols again. Well done!

josiah-liew, 2023-07-05:
Thank you for the suggestion of the factory reset. We are currently experiencing the same thing and will proceed as you have done.

yuanlee0325, 2024-02-01:
Hi! I also encountered this problem, but it has some differences. The difference is that 'it keeps cancelling and i cannot run the robot resetting function.' So, could u help me figure this issue?

[image]

cemull3, 2024-02-01:
This is a different error from what I have received, but perhaps go into devices, select the robot, select "proceed to run", cancel the protocol, and then perform a factory reset once the protocol has canceled? From the screenshot you sent, it looks like the robot still has a protocol open that it at least thinks it is running, which I'd imagine would prevent you from doing anything.

Screenshot.1.png (view on web)<https://github.com/Opentrons/opentrons/assets/158497203/3b612f24-2a62-4707-b9ba-0ed1b7553111>

-

closed 2024-02-29 by SyntaxColoring