Opentrons/opentrons #13167: bug: App cannot run on Ubuntu 22.04 LTS on WSL2 due to receiving SIGTRAP

s-wakaba, 2023-07-26:
### Overview

I tried to use Opentrons App for linux on WSL2 and connect my OT-2 machine via an ethernet tunnel over ssh because no wireress access is available in my lab and I want to use Opentrons App in front of the OT-2 machine and in my office room.

On Ubuntu 20.04 LTS on WSL2, Opentrons App for linux `Opentrons-v6.3.1-linux-b30754.AppImage` works correctly. However, on Ubuntu 22.04 LTS, the app cannot start and quit with message `Trace/breakpoint trap`.
Linux kernel is `kernel 5.15.90.1-microsoft-standard-WSL2`.

I'm not familier with AppImage debugging. but I tried to analyze the process with GDB and found that the app process received a SIGTRAP signal and there is a strange access to an address  `0xaaaaaaaaaaaaaaaa` in the backtrace.

```
$ gdb ./Opentrons-v6.3.1-linux-b30754.AppImage
(gdb) r
...
(node:522) MaxListenersExceededWarning: Possible EventEmitter memory leak detected. 11 drain listeners added to [Console]. Use emitter.setMaxListeners() to increase limit
(node:522) MaxListenersExceededWarning: Possible EventEmitter memory leak detected. 11 close listeners added to [Console]. Use emitter.setMaxListeners() to increase limit
(node:522) MaxListenersExceededWarning: Possible EventEmitter memory leak detected. 11 finish listeners added to [Console]. Use emitter.setMaxListeners() to increase limit
Thread 1 "opentrons" received signal SIGTRAP, Trace/breakpoint trap.
0x00005555580f9be5 in ?? ()
(gdb) bt
#0  0x00005555580f9be5 in  ()
#1  0xaaaaaaaaaaaaaaaa in  ()
#2  0x2d31315862696c1e in  ()
#3  0x312e6f732e626378 in  ()
#4  0x00001a540033a100 in  ()
#5  0x00007fffffffd4d0 in  ()
#6  0x000055555e529994 in  ()
#7  0x00007f0031317806 in  ()
#8  0x000055555a81199e in operator new(unsigned long) ()
```

No log files were genrated in the log directory `~/.config/Opentrons/logs/` and no GUI window appeared on the display. 

### Operating system

Linux

koji, 2024-04-06:
@s-wakaba 
Have you installed `libfuse2` before running Opentrons App?
If not, please install that on your WSL.

```shell
sudo apt install libfuse2
```

koji, 2024-04-08:
@s-wakaba 
if the above doesn't fix your issue, please contact to our support team.
support atmark opentrons.com

closed 2024-04-09 by koji