Opentrons/opentrons #13350: bug:

MSMDG, 2023-08-21:
### Overview

**### Accessing a CSV**
- Opentrons app/server: v6.3.1
- API: 2.11 or 2.14

Accessing a CSV appears to have some issues with the latest version. It appears there are 2 issues.
1) Importing a protocol
Importing a python protocol that needs access to a CSV in the "/var/lib/jupyter/notebooks" directory causes the protocol analysis to fail with the following error "list index out of range" (since it seems it does not have access to the aforementioned path). Even if the protocol analysis fails you can still run the protocol, however...
2) Labware offset data when protocol analysis fails 
If protocol analysis fails the existing labware offset data that are usually visible in the "proceed to setup" page as well as later on in "Step 3. Labware Setup, List View" are not visible with a message of "No offset data available".

### Steps to reproduce

1) CSV to use
[123_SEQE_123 (1).csv](https://github.com/Opentrons/opentrons/files/12393255/123_SEQE_123.1.csv)

2) Image illustrating issues
[image]

3) Code to reproduce

`from opentrons import protocol_api, types
import math
import csv
import glob
import subprocess

path = glob.glob('/var/lib/jupyter/notebooks/Test/*SEQE*.csv')[0]
with open (path,newline='') as csvfile:
     rows = csv.reader(csvfile)
     dna_sour = []
     dna_vol = []
     multiplication_factor = []
     sampleno = 0 #rowcount
     next (rows)
     for row in rows:
         sampleno+= 1
         a = row[1]
         dna_sour.append (a)
         b = float (row[2])
         dna_vol.append (b)
         c = float (row [3])
         multiplication_factor.append(c)`

Although not a favourable solution I managed to bypass these issues by using the code below:

`try:
 path = glob.glob('/var/lib/jupyter/notebooks/Test/*SEQE*.csv')[0]
except IndexError:
 path = glob.glob('C:/OpentronsMockCSVs/Mock_CSV_SEQE.csv')[0]`

So that it reads a CSV stored on windows for the protocol analysis but when simulating reads the correct CSV stored on var/lib/jupyter/notebooks/Test/. 

### Expected behavior

Maybe if the protocol analysis fails you could mark the protocol as ok and removes any. This in return could also enable the software to retrieve labware offset data even if protocol analysis fails.

### Operating system

Windows

### System and robot setup or anything else?

- Opentrons app/server: v6.3.1
- API: 2.11 or 2.14
- USB Connection

ClaireLee22, 2023-09-03:
I recently encountered a similar issue. While researching the problem on this webpage (https://support.opentrons.com/s/article/Protocol-analysis),  I discovered that "Protocols that used to run perfectly on a previous software version might have a "File not found error" because the file is on the OT-2 and not on the Protocol Analysis engine in the app. ".

It appears that the Protocol Analysis engine doesn't recognize the path in the OT-2 machine when you import the Python script. To address this, I've implemented a workaround using a try-except block. This allows me to handle the "File not found error" gracefully, ensuring that the previous offset data is automatically reloaded. However, there is a drawback to this approach – even if your file or path is incorrect, no error message will be displayed.

MSMDG, 2023-09-04:
Hello @ClaireLee22 I've taken a similar approach with you (see below "Code to reproduce"). You are right no error message will be displayed for the protocol analysis step, it will show an error after pressing start setup.

wolfrickw, 2023-11-07:
Try to reimport your protocol after some modification in your csv.... You'll see no modification compared to you first import, the "external" csv is cached and it is not read anymore..... or at least untill you change the csv name and modify your protocol