Opentrons/opentrons #13375: bug: Current Step shows just a "?" instead of a view of the current step (API 2.14)

MSMDG, 2023-08-24:
### Overview

Current Step shows just a "?" instead of a view of the current step (API 2.14). 
Scrolling through the run preview page does not show the current step button.
The current step is not highlighted.

[image]

### Expected behavior

Actually show the current step preview. Important when pausing but highly important when a pause command gives instructions to the robot user.

### Operating system

Windows

### System and robot setup or anything else?

- V6.3.1
- USB
- API 2.14

ecormany, 2023-08-25:
This is expected behavior in v6.3.1 of the Opentrons App for protocols that do not have a determinate number of steps. A trivial example would be something like:
```
from opentrons import protocol_api
import random

metadata = {
    'apiLevel': '2.14',
    'protocolName': 'Indeterminate step count',
}

def run(protocol: protocol_api.ProtocolContext):
    tiprack = protocol.load_labware('opentrons_96_tiprack_300ul', 1)
    pipette = protocol.load_instrument('p300_single', 'right', tip_racks=[tiprack])

    how_many = random.randrange(96)
    for _ in range(how_many):
        pipette.pick_up_tip()
        pipette.return_tip()
```
This protocol will pick up a different number of tips on each execution. 

The app should display a modal for pause steps, even when the total step count is unknown.
[image]

closed 2023-08-25 by ecormany

MSMDG, 2023-08-29:
Hello @ecormany and thank you for your comment. After lots of experimenting I found that the code below was causing the abnormal behaviour (although this was working on API 2.11):
```

if protocol.rail_lights_on==False:
    protocol.set_rail_lights(True)

```

I replaced the above with: 
```

protocol.set_rail_lights(True)

```

To replicate the issue you can run the protocol below:

```

from opentrons import protocol_api
import random

metadata = {
    'apiLevel': '2.14',
    'protocolName': 'Indeterminate step count',
}

def run(protocol: protocol_api.ProtocolContext):
    tiprack = protocol.load_labware('opentrons_96_tiprack_300ul', 1)
    pipette = protocol.load_instrument('p300_multi_gen2', mount='left', tip_racks=[tiprack])

#Lights On if Off
    if protocol.rail_lights_on==False:
        protocol.set_rail_lights(True)

    pipette.pick_up_tip()
    pipette.return_tip()

```

#13375

ecormany, 2023-08-29:
Ah yes, that's an even simpler example! The API can't predict whether that conditional will evaluate as true, because you can toggle the lights outside of a protocol. So it creates an uncertainty of ±1 step.