Opentrons/opentrons #13559: feat: Option to have location of tip drop in trash be varied

SPC-Archer, 2023-09-15:
### Overview

When running long protocols which use many tips, they can sometimes create very high piles. I haven't directly observed it directly, but I am pretty sure these piles can cause the tip to not eject properly and cause other problems with the run. 

### Implementation details

I would like a simple setting that could be set in the protocol.

Something like: `protocol.set_trash_variation(True)` 

This would then add a slight randomisation or systematic variation to where the OT2 drops a tip in the trash to avoid (or at the very least delay) a mountain of tips forming.

ecormany, 2023-09-15:
Two recent PRs address this issue, and the behavior will be available in v2.15 of the Python Protocol API.
- #12960 alternates between two locations in the trash bin by default.
- #13504 fixes a bug, so you can once again use `drop_tip` with any `Location`, if the default behavior isn't sufficient.

closed 2023-09-15 by ecormany

SPC-Archer, 2023-09-17:
Thanks! I had a quick look through the recent issues to see if it had been implemented already but not the PRs.