Opentrons/opentrons #13639: Opentrons Lagging At on Robot Anaylsis

rlenny123, 2023-09-24:
Hello, I have an OT-2 currently running 6.3.1.
I am currently experiencing an issue where protocols will take  far longer  to analyze on the robot than usual.  They can take up to 5 minutes to load a protocol with just a few lines.  It can stay stuck on analysis seemingly indefinitely  if there are multiple files on the robot and the first few analysis runs aren't cancelled and restarted. This issue has been getting worse and recently the robot has started to hang between transfers with lists of wells in them for up to 30 seconds. Deleting files  user_storage/protocol files through jupyter notebook or reinstalling the server does not help. Upgrading to 7.0.0 beta causes the app to stop communicating with the robot. What can I do? 

A similar issue arose here a few versions ago:  
https://github.com/Opentrons/opentrons/issues/2042
But it seems unrelated and was never solved anyways.

koji, 2023-10-20:
@rlenny123 
Could you reach out to our support team [support@opentrons.com](mailto:support@opentrons.com), if you still face the issue?

SyntaxColoring, 2024-02-20:
Closing, presuming this has been resolved. There have been [a number of performance improvements since v6.3.1.](https://github.com/search?q=repo%3AOpentrons%2Fopentrons+perf&type=commits&s=committer-date&o=desc)

closed 2024-02-20 by SyntaxColoring