Opentrons/opentrons #13886: bug: setting rail lights in v2

ekg, 2023-10-31:
### Overview

I am unable to set the rail lights on and off with v2.15. I'm instructed to `systemctl stop opentrons-robot-server` to use the gpio, but then issuing `protocol.set_rail_lights(True)` has no effect.

```python
import opentrons.execute
protocol = opentrons.execute.get_protocol_api('2.15')
protocol.set_rail_lights(True) # nothing happens
```

What is the correct way to manipulate the gpios from python?

### Operating system

None

ekg, 2023-12-16:
Any update on this?

SyntaxColoring, 2024-01-02:
Apologies—we haven't had a chance to look into this yet.

sfoster1, 2024-09-16:
I tried this on our upcoming 8.0.0 release with your sample code above and it worked, so it will be fixed at least in that version.