Opentrons/opentrons #13909: bug: OT-2 robot stops on the "analyzing on robot" phase for protocol running and does not progress past this for any protocols.

ELopato, 2023-11-03:
### Overview

Since installing an update to the OpenTrons App and robot yesterday, the robot no longer runs any protocols. It is connected and can be interacted with (i.e. turning on lights works), but in the app, when I am attempting to load a protocol, the app never progresses past "analyzing on robot" even after an hour of waiting. This has been tried on protocols made and used before the update as well as on some made after.

### Operating system

Windows

Avi-Til, 2023-11-06:
Same issue. Latest update to 7.0.2 released today doesn't fix it.

GBell93, 2023-11-07:
Same issue here. Version 7.0.2 stuck at "analyzing on robot" forever, no prompt to update robot software version.

Avi-Til, 2023-11-09:
> > Same issue. Latest update to 7.0.2 from 7.0.1 doesn't fix it. Also, robot server version is still in 6.3.1 with no prompt to update.
> > Update: Fixed for me. Looks like there was a delay in fetching the latest robot server version after updating the app. Updating the robot server to the same version as the app fixed it.
> 
> @Avi-Til How much of a delay did you have? Did you just leave the app open for a bit? We are still having the issue.

Around overnight. I'm not sure why there was a delay between the App update and recognizing an update for the robot as well is needed. App was closed but PC was not shutdown during the overnight period

SyntaxColoring, 2024-02-29:
Thanks both of you for the report, and sorry for the very delayed reply.

I'm going to close this because it sounds like the "stuck on 'analyzing on robot'" problem has been resolved.

The long-lived mismatch between the app and robot versions certainly is confusing, though. I'll pass that along.

closed 2024-02-29 by SyntaxColoring