Opentrons/opentrons #13938: bug:

wolfrickw, 2023-11-07:
### Overview

External data (for ex. some csv from /data/user_storage) is cached after the first import of a protocol. The protocol will use always the same imported data during the next imports, no mater that the file changed or not. To be able to use new data you have to rename your data file and modify your protocol accroding to the new file name.

Caching happens with outputs too, only the firtst protocol import will write an output file, after that no files are written, even if the previous output is deleted.
 

### Steps to reproduce

Create a file called "pp.csv" in /data/user_storage. 
Put some data in it. 
Import the protocol from below. 
Have a look on run preview (content of pp.csv is shown), notice that out.csv was created in /data/user_storage..
Change the content of pp.csv.
Reimport the protocol.
Have a look on run preview, notice that the changes in pp.csv cannot be seen.
Delete out.csv.
Reimport the protocol.
Notice that no output (out.csv) is generated.
Change the name of pp.csv to somting else. Change it in the protocol too.
Now you see the changes.

import os
from opentrons import protocol_api
metadata = {
    'protocolName': 'xxxxx',
    'author': '',
    'apiLevel': '2.9'
}

loc = os.path.dirname("/data/user_storage/")
with open(f"{loc}/" + "pp.csv", "r") as smthing:
    xx=smthing.read()
with open(f"{loc}/" + "out.csv", "w") as www:
    www.write(str(xx))

def run(ctx: protocol_api.ProtocolContext):
    ctx.comment(xx) 

### Current behavior

As described above, partial code execution (doest import data, dosnt write ouput) after the first import.

This bug (or feature???) makes OT2 extremely unflexible, and to some extent totally useless for those who are not running the same protocol with exactly the same samples over and over again...  partial, unwarned, undocumented code execution due caching is a nightmare....

### Expected behavior

No cacheing of external data. Imported data should be imported, not just pretended to be imported.  

### Operating system

Windows

### System and robot setup or anything else?

app 7.02, wifi

SyntaxColoring, 2023-11-15:
Hi, thanks for the detailed report.

I agree that this is confusing, surprising, and a change from how things worked prior to v7.0.0.

Just to be precise about what's going on, let me correct a few details. If you actually re-run the protocol, it *will* use the latest version of your external data. The *run preview* in the Opentrons App will be outdated, but the actual behavior on the robot will be up to date. Running the protocol basically `exec()`s the Python script; it doesn't cache anything. The problem is that the protocol *analysis* is being cached according to the contents of your `.py` file; the cache isn't smart enough to take your external data into account.

If you're curious about more of the technical details, you can read my comment [here](https://github.com/Opentrons/opentrons/issues/13868#issuecomment-1786013182), replying to someone else who was running into the same problem.

Anyway, yes—this is a problem and we're looking into it.

> To be able to use new data you have to rename your data file and modify your protocol accroding to the new file name.

A slightly easier workaround is to rename just the protocol file, like `my_protocol_1.py`, `my_protocol_2.py`, etc. You don't have to rename the data file or modify the contents of the protocol file. (I understand that this is still gross, though. You shouldn't have to rename anything.)

wolfrickw, 2023-11-22:
Thanks for the feedback.
I think that some caching happens since no output (out.csv) is generated (using the script from above) after the rerunning the script (if i remember correctly). You can check it by deleting the out.csv created during the  first run.
Anyway, i went back to 4.7 and everithing is fine.
I just noticed that you mentioned "prior to v7.0.0."... In versions 6.x things are "normal"? Or in which version  this feature/bug was introduced?
Thanks
F

SyntaxColoring, 2023-11-28:
> I just noticed that you mentioned "prior to v7.0.0."... In versions 6.x things are "normal"? Or in which version this feature/bug was introduced?

Correct. v6.x will reanalyze the protocol on the robot every time you upload it. This unexpected behavior was introduced in v7.0.0. (PR #12193 / commit 4bbc247aa9f91ac98f8968f1156c3575f9f9b8de.)

SyntaxColoring, 2023-11-28:
> Some off topic question:
> Is it a way to tranfer files directly from one OT2 to another? It seems that the busybox version (at least on my ot2s) does not have scp, nfs, cifs clients included. I might be wrong but this is not my field of expertise...:(

I think you're right that it doesn't have any of those clients installed, so no, I don't think there's an easy way to initiate a transfer like that from an OT-2.

The closest thing would be to do the transfer from your computer:

```shell
scp root@<robot_1_ip>:/path/to/file root@<robot_2_ip>:/path/to/file
```

wolfrickw, 2023-11-29:
Thanks.
I think including scp clients in the busybox wont be a big deal, but a nice and really useful feature....

MSMDG, 2024-02-06:
@SyntaxColoring @wolfrickw  
I have the same issue as well. Hopefully this will get fixed soon.

SyntaxColoring, 2024-03-04:
Some other people have reported this, so I've written it up as #14598 to have one central ticket. Closing this ticket in favor of that one.

Thank you again for this report!

closed 2024-03-04 by SyntaxColoring