Opentrons/opentrons #14033: bug: robot frozen when a protocol is cancelled

MurielGelin, 2023-11-21:
### Overview

when you cancel a protocol, it fails to cancel, then the only solution is to restart the frozen robot, and so the previously determined offset data are not saved !

### Expected behavior

when a protocol is cancelled, the offset data should not be lost, the robot should not freeze indefinitely

### Operating system

Linux

### System and robot setup or anything else?

v7.02

oyikuzus, 2024-01-27:
I have the same problem with a Windows PC. My student with a Mac laptop can cancel programs without any issue.

danolson1, 2024-02-07:
I have the same problem. 

There was a note that this was fixed in version 7.1.0 https://github.com/Opentrons/opentrons/pull/14215. I am running version 7.1.1, and still seem to be having this problem.

@sfoster1 Seth, do you have any ideas what might be going on?

-Dan

KU-JonasL, 2024-02-15:
I have experienced this issue/bug over my last runs. The app is trying to cancel the run, but is stopped. 
I reset it by turning off and on the OT-2
Not all runs which are cancelled, but 50% so far. It seems like my issue happens when the gantry is suppose to be active: pipetting or moving, but I have not seen this enough to be 100% sure yet.

SyntaxColoring, 2024-02-29:
If you still notice this problem after updating to v7.2.0 (which isn't out yet, at the time of writing), please provide:

* Protocol details
    * Protocol Designer or Python
    * The Python `apiLevel`, if you're using Python
    * If you're able to share it, the protocol file itself and any custom labware definitions that are required to run it
* A description of where your protocol was cancelled
    * Videos or pictures are best, if you manage to catch it

Thank you!

bktorrevillas, 2024-06-06:
Opentrons OT-2 frozen after attempt to cancel protocol during "Pause until Temp reached" step. 

App Software: 7.3.0
**Robot**: v7.3.0     **Firmware Version**  v1.1.0-25e5cea  **Supported Protocol API Versions** v2.0 - v2.18

Running protocol from library: https://protocols.opentrons.com/protocol/516336
For 24 samples, single channel left, multi channel right

Thanks for your help

DelphiVH, 2024-07-03:
Hi!

Opentrons OT-2 frozen after attempt to cancel protocol during "Blowing out at well A1 of Fixed Trash in at 92.86 µL/sec" step (run preview line 213, in "ADDING STANDARDS TO PCR tubes" part).

App Software: 7.3.1
Robot: v7.3.1 Firmware Version v1.1.0-25e5cea Supported Protocol API Versions v2.0 - v2.19

Running **python** protocol written by Opentrons support (**apiLevel 2.13**), adjusted myself and works fine (except for freezing when cancelling)

Protocol (in txt, could not attach .py file):
[Standard_curve_preparation_03072024.txt](https://github.com/user-attachments/files/16086359/Standard_curve_preparation_03072024.txt)
Custom labware definition:
[evosep_96_pcrstrip_rack.json](https://github.com/user-attachments/files/16086391/evosep_96_pcrstrip_rack.json)

Used for preparing STD curve in epps and distibuting to PCR strips.
Left pipette: single p1000
Right pipette: single p300

Thanks a lot for your help!