Opentrons/opentrons #14416: bug: Different heights when calling the same height of a labware with different pipettes (v7.0.1)

MSMDG, 2024-02-02:
### Overview

Calibrating a labware with a P20S and running a protocol with a P20 while calling the pipette to move at 1mm from the bottom will give off a correct result i.e. pipette will move at 1mm from the labware's bottom.
However, in the same protocol incorporating a P300S and calling the pipette to move at 1mm of the labware will give off a wrong result by about -1.5mm i.e. the pipette will move at -0.5 from the labware's bottom.
This causes major inconsistencies in ones protocol especially if you are working with small volumes. In addition, this means that to overcome this one should incorporate the difference in the protocol something that is not ideal at all.

### Steps to reproduce

See overview

### Operating system

Windows

### System and robot setup or anything else?

OT-2R
App/server: v7.0.1
Wi-Fi

KU-JonasL, 2024-02-19:
I experience the same bug/ a very, very similar bug.

**What happens:** 
I upload the protocol (python API) to the Opentrons App, as usual. I setup the OT-2 and go through the calibration of the off-sets, as usual. When calibrating the offset with the tip for labware, I make sure the tip of the tip is flush with the well/tube/etc. and that it is centered in the middle, as usual. It is the single pipette (and/or the lowest volume) that is used for the tip off-set position. 

I then setup up the OT-2, and begin the protocol. However, as the non-off-setting pipette try to aspirate or dispense, it want to go further down than the labware is position, aka it crashes with the bottom of the labware (the P200 tips gets bend). Beside seeing labware the pressure on the labware, I also double check if the labware is movable - it is not, aka it is pinned.
This behavior goes on for several labware types and positions, so it is not a single labware issue.

The error happens regardless if it is a p10+p50 setup or a p10+m20 setup.
The error does not seem to happen for the off-set calibrating pipette.

**Background information**
We have recently updated to 7.1.1, and it is after the update this bug is seen. All the inhouse protocols work well before.
We have recently created a web application for custom creation of inhouse protocols, however, this nothing is targeting the pipetting height here.

**Setup**
OT-2, v7.1.1
Wi-Fi
Linux PC/App, v7.1.1
Protocols: Python API, 2.16 (updated after error)

**How have I tried to solve it?**
I have tried to duplicate the error state: Same error happens despite different runs, and between shut-off states.
I have tried to recalibrate the pipettes for the OT-2(s): Bug persist
I have tried to replicate error on other OT-2 with different pipette setup and different protocol (same robot version, same PC+app): Bug persist.
I have tried to 'loosen' the height off-setting (find tip flush, then move up tip): Bug persist, though less severe problem/pinning.
I have tried to remove off-set, keeping normal setup: Bug do not happen, however with lesser accuracy.
I have tried to update the API-level in my protocols to newest version (2.16 with robot type): Bug persist.

**Thoughts**
It seem like the off-set information is carried over to the other pipette wrongly: Like it is presuming the second pipette has to work from the same position, rather than applying the off-set data to the original position of the second pipette.  
Or that the application of the off-set to the second pipette is wrong

SyntaxColoring, 2024-02-29:
Thank you both for the report.

I'm not sure if the two of you are suffering from exactly the same combination of underlying problems, but here are some likely culprits.

First: In general, any time there's a problem where one pipette works but the other doesn't, I'd take a look at pipette offset calibration and tip length calibration. (It sounds like @KU-JonasL has already done this.)

Second, and probably more importantly, we've identified a bug that causes the robot to apply tip length calibration inconsistently, which can cause all kinds of mayhem. This will be fixed in v7.2.0. (PR #14512, internal ticket RSS-468.) It's difficult to confirm if this is totally responsible for the errors you're noticing, but if you've redone tip length and pipette offset calibration and you're still seeing this, it might have something to do with it.

SyntaxColoring, 2024-02-29:
I'm going to close this because GitHub tends to not be a good venue for troubleshooting positioning and calibration problems.

If you still have this problem after we release v7.2.0, I would contact Opentrons Support. They can help discern between, say, a software bug (which we can help with here) vs. a hardware problem with your robot (which we can't).

Thanks again!

closed 2024-02-29 by SyntaxColoring