Opentrons/opentrons #14513: bug: Flex OS protocol engine not re-analyzing protocols with variable input in JN

labautom8, 2024-02-16:
### Overview

Variable CSV input is a very nice tool to write parametrized protocols on the OT Flex. For example, when using the 96 channel head, one might want a protocol to do the following:

1. Be configured for 96 channel pick up when there are 96 samples
2. Be configured for 8 channel pick up when there are < 96 samples (i.e. multiples of 8) 

To do this, one approach is to input user defined runtime variables from a CSV on the robots Jupyter notebook, and use those parameters in a protocol with some small exception handling. 

[image]

The current bug is that the above approach works as designed, however, the protocol engine on the flex does not account for the changes in parameters at the level of the Flex LCD touch display. For example if the code was something like this: 

if SampleCount = 96:
             configure for 96 channel pickup 
             handle liquid by plate
if SampleCount < 96:
             configure for 8 channel pickup 
             handle liquid by col on plate

If the user changed the number of samples from 96 to say 48 and then saves the CSV in the JN - the protocol will execute on the flex as it is programmed to do (i.e. in this case, with a 8-channel configuration), but the protocol run log (actions, comments, pauses, etc) are not updated on the display of the flex, which casts some doubt to end users that the protocol executes as designed (thought it is obvious that the flex does what it is instructed to do while watching) 

In this case, the protocol output dispalyed on the on Flex LCD remains the same as the first time the protocol was ran (in this case, for when Samples = 96), and the actions no longer correspond to what the instrument actually executes against. No errors are thrown, but each step just displays a (...) as the current executed step does not match what is being executed. 

### Steps to reproduce

1) Create a protocol with variable user input using a file hosted in JN (say sample #) 
2) Send this protocol to Flex, and run with the initial parameter set (make protocol.comment with Sample# and view before executing) 
3) Change a parameter in the variable user input file on the JN, save the NB
4) Re-execute the protocol, noting the run log of actions not being in line with opentron.simulate output, protocol.comment will display the previous sample#.

### Current behavior

The protocol reports that it is doing what it did the first time a protocol was executed. 

### Expected behavior

The protocol engine should re-analyze the protocol for changes before executing, so that the run-log displays as expected. 

Feat-option: included a re-analyze function on the Flex much like that found in the run-app so that the simulation output can be refreshed and then concurrent with user-defined changed. 

### Operating system

Windows

### System and robot setup or anything else?

v7.1.1
Eth - but operating at the LCD of Flex

vollmermvi, 2024-02-28:
@SyntaxColoring - if more details around the bug are needed - feel free to message me directly!

labautom8, 2024-02-28:
@SyntaxColoring - if more details around the bug are needed - feel free to message me directly!

SyntaxColoring, 2024-03-04:
Thank you for the detailed report, and apologies for the delayed reply.

I think you're seeing something reported by some other people, too. I've just written up #14598. See there for a description of the problem and some workarounds.

I'm closing this to try to centralize discussion over on #14598, but let me know if it looks to you like something else is going on, or if I can clarify anything.

closed 2024-03-04 by SyntaxColoring