Opentrons/opentrons #14630: bug: cancel a protocole during its run lead an infinite "cancelling" message

mumugelin, 2024-03-12:
### Overview

When I try to cancel a protocole which is running, the button "cancelling" is rolling indefinitely so that I have to switch off the robot to regain control.

### Expected behavior

When I want cancel a protocol, I should be able to regain control without switching off the robot!

### Operating system

Linux

### System and robot setup or anything else?

App v7.2.0

koji, 2024-04-21:
@mumugelin 
Could you try to use the latest app and system image?
https://github.com/Opentrons/opentrons/releases/tag/v7.2.2

mumugelin, 2024-04-22:
> [ { ***@***.***": "http://schema.org", ***@***.***": "EmailMessage",
> "name": "View Issue" }, "description": "View this Issue on GitHub",

Hi

I have just upgraded to v7.2.2 and there is still the same problem when 
cancelling a protocole: never endind "cancelling run" button !

Muriel

-- 
Dr Muriel Gelin

Research Engineer
Centre de biochimie Structurale de Montpellier
Team "ABCIS: Avanced Biology, Chemistry, Informatics Studio"

CNRS UMR 5048 - UM - INSERM U 1054
29 rue de Navacelles
34090 MONTPELLIER Cedex - FRANCE
http://www.cbs.cnrs.fr

Tel: (33) 04 67 41 77 12

--------------m44G0HqI6RvdCi0yaLX1Gm5G
Content-Type: text/html; charset="UTF-8"
Content-Transfer-Encoding: 8bit

<!DOCTYPE html>
<html>
  <head>
    <meta http-equiv="Content-Type" content="text/html; charset=UTF-8">
  </head>
  <body>
    <p>Hi</p>
    <p>I have just upgraded to v7.2.2 and there is still the same
      problem when cancelling a protocole: never endind "cancelling run"
      button !</p>
    <p>Muriel<br>
    </p>
    <div class="moz-cite-prefix">Le 21/04/2024 à 06:46, koji a écrit :<br>
    </div>
    <blockquote type="cite"
***@***.***">
      <meta http-equiv="Content-Type" content="text/html; charset=UTF-8">
      <p dir="auto"><a class="user-mention notranslate"
          data-hovercard-type="user"
          data-hovercard-url="/users/mumugelin/hovercard"
          data-octo-click="hovercard-link-click"
          data-octo-dimensions="link_type:self"
          href="https://github.com/mumugelin" ***@***.***</a><br>
        Could you try to use the latest app and system image?<br>
        <a
href="https://github.com/Opentrons/opentrons/releases/tag/v7.2.2"
          moz-do-not-send="true" class="moz-txt-link-freetext">https://github.com/Opentrons/opentrons/releases/tag/v7.2.2</a></p>
      <p
style="font-size:small;-webkit-text-size-adjust:none;color:#666;">—<br>
href="https://github.com/Opentrons/opentrons/issues/14630#issuecomment-2067902827"
          moz-do-not-send="true">view it on GitHub</a>, or <a
          moz-do-not-send="true">unsubscribe</a>.<br>
src="data:image/gif;base64,R0lGODlhAQABAJEAAP///wAAAAAAAAAAACH5BAkAAAAALAAAAAABAAEAAAgEAAEEBAA7"
alt="Web Bug from https://github.com/notifications/beacon/BG4VNCP5AJJ3HN677SLT6ATY6NABXA5CNFSM6AAAAABESGKDNKWGG33NNVSW45C7OR4XAZNMJFZXG5LFINXW23LFNZ2KUY3PNVWWK3TUL5UWJTT3IGYWW.gif"
          moz-do-not-send="true" width="1" height="1"><span
style="color: transparent; font-size: 0; display: none; visibility: hidden; overflow: hidden; opacity: 0; width: 0; height: 0; max-width: 0; max-height: 0; mso-hide: all">Message
          ID: <span>&lt;Opentrons/opentrons/issues/14630/2067902827</span><span>@</span><span>github</span><span>.</span><span>com&gt;</span></span></p>
      <mailscannerscript2496897 script="" type="application/ld+json">[
        {
        "target":
<a class="moz-txt-link-rfc2396E" href="https://github.com/Opentrons/opentrons/issues/14630#issuecomment-2067902827">"https://github.com/Opentrons/opentrons/issues/14630#issuecomment-2067902827"</a>,"url":
<a class="moz-txt-link-rfc2396E" href="https://github.com/Opentrons/opentrons/issues/14630#issuecomment-2067902827">"https://github.com/Opentrons/opentrons/issues/14630#issuecomment-2067902827"</a>,
        "name": "View Issue"
        },
        "description": "View this Issue on GitHub",
        "publisher": {
        "name": "GitHub",
        "url": <a class="moz-txt-link-rfc2396E" href="https://github.com">"https://github.com"</a>
        }
        }
        ]</mailscannerscript2496897>
    </blockquote>
    <pre class="moz-signature" cols="72">-- 
Dr Muriel Gelin

Research Engineer
Centre de biochimie Structurale de Montpellier
Team "ABCIS: Avanced Biology, Chemistry, Informatics Studio"

CNRS UMR 5048 - UM - INSERM U 1054
29 rue de Navacelles
34090 MONTPELLIER Cedex - FRANCE
<a class="moz-txt-link-freetext" href="http://www.cbs.cnrs.fr">http://www.cbs.cnrs.fr</a>

Tel: (33) 04 67 41 77 12</pre>
  </body>
</html>

--------------m44G0HqI6RvdCi0yaLX1Gm5G--

koji, 2024-04-22:
@mumugelin 
Is it possible to share your protocol?

koji, 2024-04-25:
I haven't heard any update for a few days.
I'll close this. If needed, please re-open this.

closed 2024-04-25 by koji

MurielGelin, 2024-04-26:
I still have the same problem with v7.2.2 and the problem occurs for all protocols! So I don't think it will help.
But in case, here are one of my protocols.
[crystallo-kit-to-1-SwissCi-plate_viscous.txt](https://github.com/Opentrons/opentrons/files/15126916/crystallo-kit-to-1-SwissCi-plate_viscous.txt)

SyntaxColoring, 2024-04-29:
We're unable to reproduce this. If it's happening to anyone on the latest software version (v7.2.2 at the time of writing), please:

* Provide your complete protocol files, if you haven't already.
* Let us know exactly where in the protocol the problem happens. Videos are very helpful.
* Let us know if it happens every time you cancel the protocol, or only sometimes.
* Provide your robot's [troubleshooting logs](https://opentrons.my.site.com/s/article/How-to-download-the-OT-2-logs#Downloading).

MurielGelin, 2024-05-07:
Here a video of when I try to cancel ... but it never ends !

https://github.com/Opentrons/opentrons/assets/137809780/2e8b0561-2f23-4a5f-ba47-ba00a757f0c6