Opentrons/opentrons #14744: bug: No module named 'requests'

COOOLFANS, 2024-03-28:
### Overview

Hello! I used the **'requests'** module in my program and it comes up with **no module named 'requests'.** I used SSH to install the **'requests' on ot-2,** and it shows that the installation was **successful,** but the requests library is not found in the **pip list.** 
I don't know what method to use to install the 'requests' module, please help me~
[image]
[image]
[image]

### Operating system

Windows

koji, 2024-04-03:
@COOOLFANS 
Could you check this doc's `ModuleNotFoundError`?
https://support.opentrons.com/s/article/Using-Python-packages-in-Python-API-protocols#:~:text=SSH%20into%20your%20OT-2.%20Use%20pip%20to%20install,upload%20protocols%20that%20import%20and%20use%20the%20package.

koji, 2024-04-08:
@COOOLFANS 
If you still face the same issue, please contact to our support team. support atmark opentrons.com

closed 2024-04-08 by koji