Opentrons/opentrons #14814: bug: OT2 camera not responding

ChrisTwig, 2024-04-05:
### Overview

I tried the /camera/picture endpoint and recieved.
{
  "message": "Error 1000 COMMUNICATION_ERROR (CameraException): Failed to communicate with camera",
  "errorCode": "4000"
}

running `ffmpeg -y -f video4linux2 -s 640x480 -i /dev/video0 -ss 0:0:1 -frames 1 /data/user_storage/image.jpg` gives ```[video4linux2,v4l2 @ 0x12af290] Cannot open video device /dev/video0: No such file or directory
/dev/video0: No such file or directory
```
errors

### Operating system

None

koji, 2024-04-05:
@ChrisTwig 
sorry for the inconvenience to you. Software team is working on this issue.
The fix version will be released as 7.2.2.

koji, 2024-04-10:
7.2.2 has been released. please try that.
https://github.com/Opentrons/opentrons/releases/tag/v7.2.2

closed 2024-04-10 by koji