Opentrons/opentrons #14853: bug: ImportError: cannot import name 'ParamSpec' from 'typing' for 7.2.x on Python 3.9

jkfindeisen, 2024-04-10:
### Overview

Opentrons gives an ImportError when using to simulate for version 7.2 on my Python 3.9 installation while opentrons version 7.1 doesn't give that. 

### Steps to reproduce

1. Install Python 3.9.13
2. Update pip
3. Install opentrons latest version 7.2.1 with `pip install --upgrade opentrons`
4. Execute opentrons_simulate

### Current behavior

Gives the ImportError

```
C:\Software\Python\Python39-lab-enviroment\Scripts>opentrons_simulate
Traceback (most recent call last):
  File "C:\Software\Python\Python39\lib\runpy.py", line 197, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "C:\Software\Python\Python39\lib\runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "C:\Software\Python\Python39-lab-enviroment\Scripts\opentrons_simulate.exe\__main__.py", line 4, in <module>
  File "C:\Software\Python\Python39-lab-enviroment\lib\site-packages\opentrons\__init__.py", line 9, in <module>
    from opentrons.hardware_control import (
  File "C:\Software\Python\Python39-lab-enviroment\lib\site-packages\opentrons\hardware_control\__init__.py", line 12, in <module>
    from .adapters import SynchronousAdapter
  File "C:\Software\Python\Python39-lab-enviroment\lib\site-packages\opentrons\hardware_control\adapters.py", line 6, in <module>
    from .protocols import AsyncioConfigurable
  File "C:\Software\Python\Python39-lab-enviroment\lib\site-packages\opentrons\hardware_control\protocols\__init__.py", line 6, in <module>
    from .module_provider import ModuleProvider
  File "C:\Software\Python\Python39-lab-enviroment\lib\site-packages\opentrons\hardware_control\protocols\module_provider.py", line 4, in <module>
    from ..modules import AbstractModule, ModuleModel
  File "C:\Software\Python\Python39-lab-enviroment\lib\site-packages\opentrons\hardware_control\modules\__init__.py", line 1, in <module>
    from .mod_abc import AbstractModule
  File "C:\Software\Python\Python39-lab-enviroment\lib\site-packages\opentrons\hardware_control\modules\mod_abc.py", line 11, in <module>
    from ..execution_manager import ExecutionManager
  File "C:\Software\Python\Python39-lab-enviroment\lib\site-packages\opentrons\hardware_control\execution_manager.py", line 3, in <module>
    from typing import (
ImportError: cannot import name 'ParamSpec' from 'typing' (C:\Software\Python\Python39\lib\typing.py)
```

### Expected behavior

With `pip install --upgrade opentrons==7.1.0` the execution of opentrons_simulate does not give an error but prints a message instead

```
C:\Software\Python\Python39-lab-enviroment\Scripts>opentrons_simulate
usage: opentrons_simulate [-h] [-l {debug,info,warning,error,none}] [-L CUSTOM_LABWARE_PATH] [-D [CUSTOM_DATA_PATH]] [-s CUSTOM_HARDWARE_SIMULATOR_FILE] [-d CUSTOM_DATA_FILE] [-e] [-v] [-o {runlog,nothing}] PROTOCOL
opentrons_simulate: error: the following arguments are required: PROTOCOL
```

### Operating system

Windows

jkfindeisen, 2024-04-12:
Thanks for the quick answer. I switched to Python 3.10 and it works. I also see that the Python version requirements have been adapted accordingly. https://github.com/Opentrons/opentrons/pull/14867

Python 3.9 users should be careful to only install the last version (7.1.0) that worked or switch to 3.10.

closed 2024-04-15 by SyntaxColoring

BioGeek, 2024-07-04:
Your support article [Simulating OT-2 protocols on your computer](https://support.opentrons.com/s/article/Simulating-OT-2-protocols-on-your-computer) currently still has the line:

> **Update:** Python 3.10 currently **does not** allow for protocol simulation. It is recommended to use a version between 3.7.0 and 3.9.9.

So I installed python 3.9.9, got this ParamSpec ImportError and had to go install Python 3.10 to get it working. Please update your documentation.

ecormany, 2024-07-11:
Thanks for drawing our attention to the support article. We've updated it to specify Python 3.10.