Opentrons/opentrons #15110: bug: installing packages from PyPi is broken upon upgrading from v7.1.0 to v7.2.2

ahmadomira, 2024-05-07:
### Overview

After updating our OT-2 robot's software from v7.1.0 to v7.2.2, installing a Python package (from outside Python's standard library) using `pip` doesn't work. The installed packages do exist in Python's site-packages directory, yet importing them inside the Python interpreter results a `Module Not Found` error. 

Please note that this problem goes away when downgrading the software to v7.1.0. 

### Steps to reproduce

**Note:** the following problem can be reproduced by installing any PyPi package. The packages are installed following [this](https://support.opentrons.com/s/article/Using-Python-packages-in-Python-API-protocols) Opentrons tutorial.
 
After completing the initial [SSH-setup](https://support.opentrons.com/s/article/Setting-up-SSH-access-to-your-OT-2):
1. connect the robot to our local network
2. open a Powershell window, and do:
```bash
ssh -i ot2_ssh_key root@OUR_ROBOT_ID
```
3. after successful login to the robot:
```bash
pip3 install openpyxl
```
4. after a successful package installation, restart the robot. 
5. login again into the robot over SSH (repeat Step 1)
6. doing, for example:
```bash
python3
>>> import openpyxl
```

### Current behavior

 Doing, for example, this:
```bash
python3
>>> import openpyxl
```
outputs: `Module Not Found Error`

or simply attempting to uninstall the same package, with no further steps in between (as shown in the screenshot below):
```bash 
pip3 uninstall openpyxl
```
outputs: `WARNING: Skipping openpyxl as it is not installed`. 

We have tried to install packages from PyPi into Python virtual environemts ([venv](https://docs.python.org/3/tutorial/venv.html)), but the problem persists. 

### Problem Screenshot
[image]

### The installed packages do exist in Python's system directory
Navigating to Python's site-packages directory, after package installation, shows that the packages exist within the Python directory. Doing in the robots terminal, for example: 
```bash
which python3
```
outputs: `/usr/bin/python3`

Now navigating to `/usr/bin/python3/.../site-packages` shows that the installed package (and its submodules) exists, as shown in these screenshots:

[image]
[image]
[image]

### Expected behavior

Before upgrading the software, which we had to do because of other problem, we used to install our Python packages from within our Jupyter notebooks by doing:
```python
!pip install openpyxl
``` 

then import `openpyxl` in next cells with no issues. Now, doing the same results `Module Not Found` errors

### Operating system

Windows

### System and robot setup or anything else?

+ Opentrons software version: v7.2.2
+ The robot is connected to the PC via Ethernet
+ Note: the `PYTHONPATH` environment variable seems to be empty. We were not able to set it otherwise

aschleck, 2024-09-03:
Better solution: run the venv's pip with `PIP_CONFIG_FILE=/dev/null` like `PIP_CONFIG_FILE=/dev/null pip install pandas`.

The breakage on Flex is caused by https://github.com/Opentrons/oe-core/blob/2ad1ac910d393c84b7f363ae3cca23786d290935/layers/meta-opentrons/recipes-devtools/python3-pip/files/pip.conf#L1

aschleck, 2024-09-11:
@vegano1 any thoughts on this?

My guess is that OT doesn't run the Jupyter notebook inside of a venv, so if the user does `!pip install pandas` then it would by default try to install into the read-only root. So then this change forces pip to install into this particular path.

Personally I think a better way to support `!pip install pandas` in the notebook would have been to just run Jupyter inside of a venv and not force every pip invocation to use this path. But I may not understand the problem or goal fully.

One reason my approach, using venvs, is better is because with venvs you can install the latest version of Pandas. Whereas even with this `pip.conf` hack, pandas will still try to install into `/usr/bin`. If you use `pip install` with venvs then Pandas will correctly install the binary into a `bin` folder in the venv instead.

jjmc0000, 2025-01-15:
Was this ever fixed properly? I am also experiencing this issue currently