Opentrons/opentrons #15138: bug: "Override path to python" not working

steakhouserodriguez, 2024-05-08:
### Overview

The setting has no effect, even after restarting the app and computer.

### Steps to reproduce

[image]

```
#otpython.py
metadata = {
    "protocolName": "otpython",
    "apiLevel": "2.17",
}

import sys
raise Exception(sys.executable)

def run(ctx):
    pass

```

### Current behavior

protocol analysis should use my interpreter, not the bundled.

[image]

### Expected behavior

Should be 
```
/opt/miniconda3/envs/ivynatal_lab/bin/python3
```

### Operating system

Mac

### System and robot setup or anything else?

I'm using miniconda? 
App version 7.2.2

koji, 2024-07-02:
@steakhouserodriguez 
Could you check one thing?

There is the config file of the app and the following is the file path.

Library/Application\ Support/Opentrons/config.json

steakhouserodriguez, 2024-07-02:
yeah, config.json has the path I submitted. I've just emailed support@.

koji, 2024-07-13:
@steakhouserodriguez 

if you type `which python`, is the output path is the same as the path you submitted?

steakhouserodriguez, 2024-07-15:
Hmm, I've ended up setting the environment to `envs/ivynatal_lab` instead of `envs/ivynatal_lab/bin` and it seems to work now. I'm 75% sure I did try this in the past without success, but I can't reproduce the issue anymore.

> @steakhouserodriguez
> 
> if you type `which python`, is the output path is the same as the path you submitted?

I'm not sure if its relevant at this point, but the output paths would correspond to the environment's python path. I don't have python installed in system path.

koji, 2024-07-15:
@steakhouserodriguez 
Thank you for the update.

closed 2024-10-18 by koji