Opentrons/opentrons #15274: bug: Loading protocol Analyzing On Robot

haianhle, 2024-05-28:
### Overview

I have been having this issue with loading protocol, where upon trying to set up, the script kept getting stuck at "Analyzing on Robot" without anything happening. I have uninstalled and reinstalled software, reset system to factory setting and recalibrated everything, still same issue. This happened after we swapped out the 8-channel P300. I re-installed the single channel P300 and recalibrated. Instrument shows Calibration ready but protocol still cannot be loaded. Please help!

### Current behavior

Analyzing on Robot and protocol not loading.

### Expected behavior

App should loaded

### Operating system

None

### System and robot setup or anything else?

App version is 7.2. System has nothing else installed. Robot is connected via USB.

haianhle, 2024-05-28:
[image]

koji, 2024-05-29:
@haianhle 
Could you share the protocol you used?

okomagata, 2024-06-06:
I am experiencing the same issue with my setup. The protocol fails to load and gets stuck at the "Analyzing on Robot" step without any progress. This occurred after I replaced the single-channel P300 pipette. I have tried uninstalling and reinstalling the software (version 7.3), resetting the system to factory settings, and recalibrating, but the problem persists. The instrument shows that it is calibration ready, but the protocol still cannot be loaded. Please advise on how to resolve this issue.

My typical simple script for test is below.

----
from opentrons import protocol_api

metadata = {
    "apiLevel": "2.13"
}

def run(protocol: protocol_api.ProtocolContext):

    tr1_20 = protocol.load_labware('opentrons_96_tiprack_20ul', '6')
    tr1_300 = protocol.load_labware('opentrons_96_tiprack_300ul', '5')

    p20s = protocol.load_instrument('p20_single_gen2', 'right', tip_racks=[tr1_20])
    p300s = protocol.load_instrument('p300_single_gen2', 'left', tip_racks=[tr1_300])

    p20s.pick_up_tip()
    p300s.pick_up_tip()

    p20s.return_tip()
    p300s.return_tip()

koji, 2024-06-06:
@okomagata 
I tried the protocol you shared but I didn't see any issue on my end.

Do you use app v7.3 and robot system v7.3?
Could you share your robot system logs? (Robot Settings  -> Advanced tab -> Download logs?

okomagata, 2024-06-07:
Thank you, ＠koji!

Finally, I realized that the robot's system file upgrade had not been entirely completed, which solved my problem. The OT-2 resumed normal operation after a thorough system update.

In my situation, when the "Analyzing On Robot" state did not complete after starting up the robot, certain tasks, such as upgrading the robot's system files, became unavailable. Even if the protocol is stopped or the loaded software is removed from the Protocols menu, the "Analyzing On Robot" condition persists. I attempted to establish as many settings as possible and update the robot system files immediately after turning on the power, allowing me to complete this work.

Thank you very much for your valuable suggestions.

koji, 2024-06-07:
hi @okomagata 
Thank you for your response and glad to hear that you could solve the issue.

koji, 2024-06-07:
@haianhle 
I've not received any response from you for a while.
I'm going to close this issue. If you are still facing the same issue, please contact to our support team.
support(at mark)opentrons.com

closed 2024-06-07 by koji

haianhle, 2024-06-07:
@koji  I apologize for my late response. Thank you @okomagata it's exactly the problem we have it turned out. I did the manual robot file upgrade and everything is working now! Thank you both again very much :)