Opentrons/opentrons #15292: bug: Liquid sucked into 96 channel pipette head after cancellation or collision detection

HannahHuckstep, 2024-05-31:
### Overview

I am on a Flex with a 96 channel head. 
I have found repeatedly that if there is liquid in the pipette tips when the run is cancelled or a collision is detected, the robot will home itself and reset the pipette drawing liquid up into the pipette head. There is no way to stop it. 
I also usually re-rack my tips after a run, and right before they are dropped in the rack the same behavior happens where liquid is sucked into the head. 

### Steps to reproduce

Write a protocol to aspirate some amount of liquid, I recommend a small volume as to not get too much in the head (20 µl in a 50µl head should be enough). Cancel the run while there is liquid in the tips. 

### Current behavior

**Liquid drawn into the pipette head.** 

### Expected behavior

Wait until after the tips are dropped off to reset the pipette. 

### Operating system

Mac

### System and robot setup or anything else?

App: 7.2.2 
Flex: 7.2.2
Connected via Wifi

cemull3, 2025-03-11:
Has this ever been addressed? I am having the same issue running 8.1.0. This is a major problem-hoping I haven't permanently damaged anything.

patrickdiep, 2025-06-13:
How has this issue not been addressed yet?

cemull3, 2025-06-14:
> How has this issue not been addressed yet?

Hi Patrick! As far as I can tell, Opentrons does not actively engage with the content on this page. However, emailing technical support and cc'ing your sales rep will likely provide more results. 

From what I understand, this issue has been addressed in version 8.3.2. I haven't observed it since upgrading, but also I haven't been canceling protocols lately with solvent in the tip. I have noticed that if there is any solvent remaining in the tip after an attempted full dispense, the pipette will over-aspirate before a new action, such as discarding the tips or aspirating more solvent. As a work-around, I never let the robot think the pipette is empty. I aspirate an extra 1ul of air, which I never dispense. Of course, this means I can't use the full pipette capacity or blow out feature, but I'm writing my protocols in python and have created my own custom pseudo-blow out which is working great. 

Hope this helps! Again, I'd definitely recommend emailing Opentrons about issues you are observing, especially if it has caused damage to your pipette head - they are much more responsive by email.

HannahHuckstep, 2025-06-30:
It is absolutely insane that this has not been addressed. Our lab is not using the opentrons app but are using the latest API version and we just saw this behavior again last week. I've emailed opentrons again today - hope to get an answer soon.