Opentrons/opentrons #15638: feat: Calibrating from the Bottom of the labware

MSMDG, 2024-07-12:
### Overview

Calibrating from the bottom of the labware is more convenient and more precise for the user. Additionally, when transferring protocols to other OT-2 robots, calibrating from the bottom is more consistent and therefore changes to protocol heights are not necessary.
Maybe having the option to choose whether to calibrate from the top or bottom of the labware would be best.

Koeng101, 2024-09-13:
#11763

They aren't really interested in adding it back, unfortunately. So I stay downgraded to 4.7.0. It's from 2021, but honestly better software for running OT2s because it has calibrate to bottom

MSMDG, 2024-09-16:
Hi @Koeng101 yeah I wished they would bring it back, it is much easier and accurate. For now we are calibrating from the top while using a very thin piece of paper (~0.2mm thick) to have a reference and stay consistent when transferring protocols to other robots.