Opentrons/opentrons #16137: bug: Is this problem relevant to the mismatch of the firmware between ot2 robot and the heatershaker module

yuanlee0325, 2024-08-27:
### Overview

Issues with the control of HeaterShaker module.
I try to write a protocol to test the heater shaker. It can successfully trigger the shaker function at the setting speed, but it can not stop it at the setting time. But I could control it successfully by using the opentrons app.
Heatershaker moduleGen1: V1.0.4, HSV012023040409 

OT2:OT2 CEP20220901R07 

Robot Firmware version: V1.1.0-25e5cea 

### Steps to reproduce

The coding is like this:
from opentrons import protocol_api 
import opentrons.execute# Metadata 
metadata = { 
    'protocolName': 'Heater-Shaker Basic Usage', 
    'author': 'Your Name', 
    'description': 'A simple protocol to demonstrate Heater-Shaker usage with the Opentrons robot.', 
    'apiLevel': '2.13' 
} 

protocol = opentrons.execute.get_protocol_api('2.15') 

def run(): 
    global protocol 
    # Load Heater-Shaker module 
    heater_shaker = protocol.load_module('heaterShakerModuleV1', '1') 
    heater_shaker_plate = heater_shaker.load_labware('opentrons_96_deep_well_adapter_nest_wellplate_2ml_deep')    # Set Heater-Shaker module parameters 
    print(heater_shaker.serial_number) 
    print(heater_shaker.type) 
    heater_shaker.close_labware_latch() 
    heater_shaker.set_and_wait_for_shake_speed(500)         
  protocol.delay(minutes=0.5)        
  heater_shaker.deactivate_shaker() 
   heater_shaker.deactivate_heater() 

### Current behavior

1. It can successfully give the module type and serial number of the heatershaker module.
[image]
2. But it can not stop the shaker , and gaive an 'MS S500' error and M105 or M123 subsequently.

### Expected behavior

I want to write my own protocol to control the heatershaker module and then combine it with the liquid handling task.

### Operating system

Windows

### System and robot setup or anything else?

App version: 7.5.0
Robot server: 7.5.0

ecormany, 2024-08-27:
This is expected behavior when controlling modules in Jupyter notebook. You need to [stop the robot server](https://docs.opentrons.com/v2/new_advanced_running.html#using-modules) to avoid the module receiving conflicting commands, which cause this sort of error.

yuanlee0325, 2024-09-01:
> Caution: This email originated from outside of the University. Do not click links or open attachments unless you recognise the source of this email and know the content is safe. Check sender address, hover over URLs and don't open suspicious email attachments.

Hope this email finds you well.

Many thanks for your help. It is my mistake that I have not read the api carefully.

Best regards,
Yuan

ecormany, 2024-09-03:
No worries! You can reopen this issue or contact Opentrons Support if you're still having problems.

closed 2024-09-03 by ecormany