Opentrons/opentrons #16475: bug: Robot

0ont, 2024-10-14:
### Overview

When running any protocol or even just the home gantry command, the robots starts to align the position of the pipettes above the trash bin, then just crashes / looses power and automatically reboots. video attached. it can be clearly seen when the blue led goes silent. (Opentrons OT2)

https://github.com/user-attachments/assets/6e7d0cf1-1abc-479f-8239-3c593cfa1311

### Steps to reproduce

Run home gantry command.

### Current behavior

The robot crashes and reboots when I try to start any protocol, or even just run the home gantry command (Opentrons OT-2)

### Expected behavior

The home gantry command or whole protocols should execute, instead of the robot crashing and rebooting.

### Operating system

Windows

### System and robot setup or anything else?

Ethernet cable connection

Parasiterex, 2024-11-07:
I am experiencing the same thing combined with an error in the opentrons app 

This robot's API server is not responding correctly to requests at IP address 169.254.90.58

Did you find any fixes thus far?

tdlong, 2025-02-02:
We are having the same problem with an OT-2 (2018/25/26).  Even if I run the check calibration command the OT-2 homes OK, but then moves very quickly to roughly position 9, then stops and loses the connect.  I tried backdating the software to version 7.5 and the problem continues.  Did anyone find a fix??

tdlong, 2025-02-03:
There is some weirdness in the server.log file, given that the robot seems to drop the connection to my mac, and powers off, perhaps the actual event that kills everything is not recorded.

Feb 02 23:14:04.485181 249c0a uvicorn[248]:  - "GET /robot/lights HTTP/1.1" 200
Feb 02 23:14:04.529646 249c0a uvicorn[248]: BEGIN (implicit)
Feb 02 23:14:04.531798 249c0a uvicorn[248]: BEGIN
Feb 02 23:14:04.533589 249c0a uvicorn[248]: [raw sql] ()
Feb 02 23:14:04.559709 249c0a uvicorn[248]: SELECT run.id, run.protocol_id, run.created_at 
                                            FROM run ORDER BY "_ROWID_" ASC
Feb 02 23:14:04.561474 249c0a uvicorn[248]: [generated in 0.00232s] ()
Feb 02 23:14:04.579459 249c0a uvicorn[248]: SELECT action.id, action.created_at, action.action_type, action.run_id 
                                            FROM action ORDER BY "_ROWID_" ASC
Feb 02 23:14:04.581226 249c0a uvicorn[248]: [generated in 0.00216s] ()
Feb 02 23:14:04.588428 249c0a uvicorn[248]: COMMIT
Feb 02 23:14:04.633159 249c0a uvicorn[248]:  - "GET /modules HTTP/1.1" 200
Feb 02 23:14:04.637058 249c0a uvicorn[248]:  - "GET /runs HTTP/1.1" 200
Feb 02 23:14:04.806265 249c0a uvicorn[248]:  - "GET /networking/status HTTP/1.1" 200
Feb 02 23:14:05.470165 249c0a uvicorn[248]: Error response: 500 - LegacyError - KeyError: ''
Feb 02 23:14:05.472315 249c0a uvicorn[248]:  - "GET /settings/pipettes HTTP/1.1" 500
Feb 02 23:14:05.479382 249c0a uvicorn[248]: Exception in ASGI application
                                            Traceback (most recent call last):
                                              File "/usr/lib/python3.10/site-packages/uvicorn/protocols/http/httptools_impl.py", line 419, in run_asgi
                                              File "/usr/lib/python3.10/site-packages/uvicorn/middleware/proxy_headers.py", line 84, in __call__
                                              File "/usr/lib/python3.10/site-packages/fastapi/applications.py", line 290, in __call__
                                              File "/usr/lib/python3.10/site-packages/starlette/applications.py", line 122, in __call__
                                              File "/usr/lib/python3.10/site-packages/starlette/middleware/errors.py", line 184, in __call__
                                              File "/usr/lib/python3.10/site-packages/starlette/middleware/errors.py", line 162, in __call__
                                              File "/usr/lib/python3.10/site-packages/starlette/middleware/cors.py", line 83, in __call__
                                              File "/usr/lib/python3.10/site-packages/starlette/middleware/exceptions.py", line 79, in __call__
                                              File "/usr/lib/python3.10/site-packages/starlette/middleware/exceptions.py", line 68, in __call__
                                              File "/usr/lib/python3.10/site-packages/fastapi/middleware/asyncexitstack.py", line 20, in __call__
                                              File "/usr/lib/python3.10/site-packages/fastapi/middleware/asyncexitstack.py", line 17, in __call__
                                              File "/usr/lib/python3.10/site-packages/starlette/routing.py", line 718, in __call__
                                              File "/usr/lib/python3.10/site-packages/starlette/routing.py", line 276, in handle
                                              File "/usr/lib/python3.10/site-packages/starlette/routing.py", line 66, in app
                                              File "/usr/lib/python3.10/site-packages/fastapi/routing.py", line 241, in app
                                              File "/usr/lib/python3.10/site-packages/fastapi/routing.py", line 167, in run_endpoint_function
                                              File "/usr/lib/python3.10/site-packages/robot_server/service/legacy/routers/settings.py", line 326, in get_pipette_settings
                                              File "/usr/lib/python3.10/site-packages/robot_server/service/legacy/routers/settings.py", line 435, in _pipette_settings_from_known_id
                                              File "/usr/lib/python3.10/site-packages/opentrons_shared_data/pipette/mutable_configurations.py", line 276, in list_mutable_configs
                                            KeyError: ''
Feb 02 23:14:06.152504 249c0a uvicorn[248]:  - "GET /runs HTTP/1.1" 200
Feb 02 23:14:07.518499 249c0a uvicorn[248]: Error response: 500 - LegacyError - KeyError: ''
Feb 02 23:14:07.520624 249c0a uvicorn[248]:  - "GET /settings/pipettes HTTP/1.1" 500
Feb 02 23:14:07.527625 249c0a uvicorn[248]: Exception in ASGI application
                                            Traceback (most recent call last):
                                              File "/usr/lib/python3.10/site-packages/uvicorn/protocols/http/httptools_impl.py", line 419, in run_asgi
                                              File "/usr/lib/python3.10/site-packages/uvicorn/middleware/proxy_headers.py", line 84, in __call__
                                              File "/usr/lib/python3.10/site-packages/fastapi/applications.py", line 290, in __call__
                                              File "/usr/lib/python3.10/site-packages/starlette/applications.py", line 122, in __call__
                                              File "/usr/lib/python3.10/site-packages/starlette/middleware/errors.py", line 184, in __call__
[... cut ...]
                                              File "/usr/lib/python3.10/site-packages/robot_server/service/legacy/routers/settings.py", line 326, in get_pipette_settings
                                              File "/usr/lib/python3.10/site-packages/robot_server/service/legacy/routers/settings.py", line 435, in _pipette_settings_from_known_id
                                              File "/usr/lib/python3.10/site-packages/opentrons_shared_data/pipette/mutable_configurations.py", line 276, in list_mutable_configs
                                            KeyError: ''