Opentrons/opentrons #17084: bug: No module named 'opentrons_hardware'

angelovangel, 2024-12-11:
### Overview

Hi,
I cannot run Flex protocols with `opentrons_simulate`, while OT-2 protocols run fine.
I am on v8.2.0. The error is:
```
Traceback (most recent call last):
  File "/Users/angeloas/miniforge3/lib/python3.10/runpy.py", line 196, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/Users/angeloas/miniforge3/lib/python3.10/runpy.py", line 86, in _run_code
    exec(code, run_globals)
  File "/Users/angeloas/miniforge3/lib/python3.10/site-packages/opentrons/simulate.py", line 1061, in <module>
    sys.exit(main())
  File "/Users/angeloas/miniforge3/lib/python3.10/site-packages/opentrons/simulate.py", line 1020, in main
    runlog, maybe_bundle = simulate(
  File "/Users/angeloas/miniforge3/lib/python3.10/site-packages/opentrons/simulate.py", line 555, in simulate
    with _make_hardware_simulator_cm(
  File "/Users/angeloas/miniforge3/lib/python3.10/contextlib.py", line 135, in __enter__
    return next(self.gen)
  File "/Users/angeloas/miniforge3/lib/python3.10/site-packages/opentrons/simulate.py", line 377, in _make_hardware_simulator_cm
    result = _make_hardware_simulator(override=None, robot_type=robot_type)
  File "/Users/angeloas/miniforge3/lib/python3.10/site-packages/opentrons/simulate.py", line 350, in _make_hardware_simulator
    from opentrons.hardware_control.ot3api import OT3API
  File "/Users/angeloas/miniforge3/lib/python3.10/site-packages/opentrons/hardware_control/ot3api.py", line 143, in <module>
    from .backends.flex_protocol import FlexBackend
  File "/Users/angeloas/miniforge3/lib/python3.10/site-packages/opentrons/hardware_control/backends/flex_protocol.py", line 41, in <module>
    from opentrons_hardware.firmware_bindings.constants import SensorId
ModuleNotFoundError: No module named 'opentrons_hardware'
``` 

### Operating system

Mac

koji, 2024-12-14:
can you try the following command and run the simulator again?

```shell
python -m pip install 'opentrons-hardware @ git+https://github.com/Opentrons/opentrons@fc8f7113c682cd6ce48273cf7c47a242b9272cfc#subdirectory=hardware'
```

angelovangel, 2024-12-14:
Thanks, this gives git clone error:
```
Collecting opentrons-hardware@ git+https://github.com/Opentrons/opentrons@fc8f7113c682cd6ce48273cf7c47a242b9272cfc#subdirectory=hardware
  Cloning https://github.com/Opentrons/opentrons (to revision fc8f7113c682cd6ce48273cf7c47a242b9272cfc) to /private/var/folders/cy/xfqdqgvs2mn43hxw_ldk_mfslvntxp/T/pip-install-h9o6c650/opentrons-hardware_333ae0206ec143af8783cf502bba49d3
  Running command git clone --filter=blob:none --quiet https://github.com/Opentrons/opentrons /private/var/folders/cy/xfqdqgvs2mn43hxw_ldk_mfslvntxp/T/pip-install-h9o6c650/opentrons-hardware_333ae0206ec143af8783cf502bba49d3
  error: RPC failed; curl 92 HTTP/2 stream 3 was not closed cleanly: CANCEL (err 8)
  error: 2629 bytes of body are still expected
  fetch-pack: unexpected disconnect while reading sideband packet
  fatal: early EOF
  fatal: fetch-pack: invalid index-pack output
  fatal: could not fetch 74cae9c8e895118c0c80dc3e6eda6e6daeb15537 from promisor remote
  warning: Clone succeeded, but checkout failed.
  You can inspect what was checked out with 'git status'
  and retry with 'git restore --source=HEAD :/'
```

koji, 2024-12-16:
@angelovangel 
From the error, it seems that you are having a network connection issue.

angelovangel, 2024-12-16:
That fixed it, thanks!

closed 2024-12-16 by angelovangel

koji, 2024-12-16:
@angelovangel 
glad to hear that.