Opentrons/opentrons #17146: feat: Flex compatability with SSH format ed25519

radmanahmadi, 2024-12-19:
### Overview

As of Openssh v9.5, ssh-keygen will create a ed25519 key pair by default.
https://www.openssh.com/txt/release-9.5

The POST command on the FLEX SSH setup page seems to only work with RSA keys (tested on v8.1.0). 
https://support.opentrons.com/s/article/Set-up-SSH-access-to-Opentrons-Flex

The command should accept ed25519 keys. Failing that, the documentation should be updated to ask the user to specifically create an RSA key.

jwwojak, 2025-01-28:
Please see the Flex Instruction Manual section, [Command Line Operation Over SSH](https://insights.opentrons.com/hubfs/Products/Flex/Opentrons%20Flex%20manual%20REV3.pdf#page=156).

ecormany, 2025-03-12:
We've also updated the support page to indicate that you need to add the `-t ecdsa` flag when creating your key.

closed 2025-03-12 by ecormany