Opentrons/opentrons #17391: bug: opentrons_sumulate error

tobiasko, 2025-01-31:
### Overview

The `opentrons_simulate` executable gives an error when trying to run a python protocol for `{"apiLevel" : "2.21", "robotType" : "Flex"}`. Adding custom labware by `-L` doesn't help. Also the rather simple tutorial code doesn't run.

**Can't attach any .py files below!**

### Steps to reproduce

```
opentrons_simulate protocols/enrich-iST_96x_HT/enrich-iST_beadPrep_enrich_lyse_8channel_EppendorfDeepWellPlate_96_500_uL.py
Traceback (most recent call last):
  File "/Users/tobiaskockmann/PycharmProjects/opentronsflex/.venv/bin/opentrons_simulate", line 8, in <module>
    sys.exit(main())
             ^^^^^^
  File "/Users/tobiaskockmann/PycharmProjects/opentronsflex/.venv/lib/python3.12/site-packages/opentrons/simulate.py", line 1020, in main
    runlog, maybe_bundle = simulate(
                           ^^^^^^^^^
  File "/Users/tobiaskockmann/PycharmProjects/opentronsflex/.venv/lib/python3.12/site-packages/opentrons/simulate.py", line 555, in simulate
    with _make_hardware_simulator_cm(
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Library/Frameworks/Python.framework/Versions/3.12/lib/python3.12/contextlib.py", line 137, in __enter__
    return next(self.gen)
           ^^^^^^^^^^^^^^
  File "/Users/tobiaskockmann/PycharmProjects/opentronsflex/.venv/lib/python3.12/site-packages/opentrons/simulate.py", line 377, in _make_hardware_simulator_cm
    result = _make_hardware_simulator(override=None, robot_type=robot_type)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/tobiaskockmann/PycharmProjects/opentronsflex/.venv/lib/python3.12/site-packages/opentrons/simulate.py", line 350, in _make_hardware_simulator
    from opentrons.hardware_control.ot3api import OT3API
  File "/Users/tobiaskockmann/PycharmProjects/opentronsflex/.venv/lib/python3.12/site-packages/opentrons/hardware_control/ot3api.py", line 143, in <module>
    from .backends.flex_protocol import FlexBackend
  File "/Users/tobiaskockmann/PycharmProjects/opentronsflex/.venv/lib/python3.12/site-packages/opentrons/hardware_control/backends/flex_protocol.py", line 41, in <module>
    from opentrons_hardware.firmware_bindings.constants import SensorId
ModuleNotFoundError: No module named 'opentrons_hardware'
```

### Operating system

Mac

### System and robot setup or anything else?

```
opentrons_simulate --version
opentrons_simulate 8.2.0
```

koji, 2025-02-03:
can you try the following command and run opentrons_simulate again?

```shell
pip install --no-cache-dir --force-reinstall opentrons-hardware@git+https://github.com/Opentrons/opentrons@release#subdirectory=hardware
```

tobiasko, 2025-02-03:
```
(.venv) ~/PycharmProjects/opentronsflex git:[main]
pip install --no-cache-dir --force-reinstall opentrons-hardware@git+https://github.com/Opentrons/opentrons@release#subdirectory=hardware

Collecting opentrons-hardware@ git+https://github.com/Opentrons/opentrons@release#subdirectory=hardware
  Cloning https://github.com/Opentrons/opentrons (to revision release) to /private/var/folders/_l/psxm9vkj6m11z1vws9plwj6h0000gn/T/pip-install-r7h5crpc/opentrons-hardware_15a62e5c457c4700bf8f8700d8ae8a08
  Running command git clone --filter=blob:none --quiet https://github.com/Opentrons/opentrons /private/var/folders/_l/psxm9vkj6m11z1vws9plwj6h0000gn/T/pip-install-r7h5crpc/opentrons-hardware_15a62e5c457c4700bf8f8700d8ae8a08
  Running command git checkout -b release --track origin/release
  Switched to a new branch 'release'
  branch 'release' set up to track 'origin/release'.
  Resolved https://github.com/Opentrons/opentrons to commit 453cd4776074263bb990f28afdb194e9928bdbf4
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... done
Collecting pyserial==3.5 (from opentrons-hardware@ git+https://github.com/Opentrons/opentrons@release#subdirectory=hardware)
  Obtaining dependency information for pyserial==3.5 from https://files.pythonhosted.org/packages/07/bc/587a445451b253b285629263eb51c2d8e9bcea4fc97826266d186f96f558/pyserial-3.5-py2.py3-none-any.whl.metadata
  Downloading pyserial-3.5-py2.py3-none-any.whl.metadata (1.6 kB)
Collecting opentrons_shared_data==8.2.0 (from opentrons-hardware@ git+https://github.com/Opentrons/opentrons@release#subdirectory=hardware)
  Obtaining dependency information for opentrons_shared_data==8.2.0 from https://files.pythonhosted.org/packages/76/03/559eefa8b843fe4264c2f17f8ac3d3f5f69e963f2dc1b1d827db7665718c/opentrons_shared_data-8.2.0-py2.py3-none-any.whl.metadata
  Downloading opentrons_shared_data-8.2.0-py2.py3-none-any.whl.metadata (1.2 kB)
Collecting jsonschema<4.18.0,>=3.0.1 (from opentrons_shared_data==8.2.0->opentrons-hardware@ git+https://github.com/Opentrons/opentrons@release#subdirectory=hardware)
  Obtaining dependency information for jsonschema<4.18.0,>=3.0.1 from https://files.pythonhosted.org/packages/c1/97/c698bd9350f307daad79dd740806e1a59becd693bd11443a0f531e3229b3/jsonschema-4.17.3-py3-none-any.whl.metadata
  Downloading jsonschema-4.17.3-py3-none-any.whl.metadata (7.9 kB)
Collecting typing-extensions<5,>=4.0.0 (from opentrons_shared_data==8.2.0->opentrons-hardware@ git+https://github.com/Opentrons/opentrons@release#subdirectory=hardware)
  Obtaining dependency information for typing-extensions<5,>=4.0.0 from https://files.pythonhosted.org/packages/26/9f/ad63fc0248c5379346306f8668cda6e2e2e9c95e01216d2b8ffd9ff037d0/typing_extensions-4.12.2-py3-none-any.whl.metadata
  Downloading typing_extensions-4.12.2-py3-none-any.whl.metadata (3.0 kB)
Collecting pydantic<2.0.0,>=1.10.9 (from opentrons_shared_data==8.2.0->opentrons-hardware@ git+https://github.com/Opentrons/opentrons@release#subdirectory=hardware)
  Obtaining dependency information for pydantic<2.0.0,>=1.10.9 from https://files.pythonhosted.org/packages/7b/d3/75a00f07e247b615002f9423d426191785850ff1bf947803f21b6bce952b/pydantic-1.10.21-cp312-cp312-macosx_11_0_arm64.whl.metadata
  Downloading pydantic-1.10.21-cp312-cp312-macosx_11_0_arm64.whl.metadata (153 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 153.9/153.9 kB 5.7 MB/s eta 0:00:00
Collecting attrs>=17.4.0 (from jsonschema<4.18.0,>=3.0.1->opentrons_shared_data==8.2.0->opentrons-hardware@ git+https://github.com/Opentrons/opentrons@release#subdirectory=hardware)
  Obtaining dependency information for attrs>=17.4.0 from https://files.pythonhosted.org/packages/fc/30/d4986a882011f9df997a55e6becd864812ccfcd821d64aac8570ee39f719/attrs-25.1.0-py3-none-any.whl.metadata
  Downloading attrs-25.1.0-py3-none-any.whl.metadata (10 kB)
Collecting pyrsistent!=0.17.0,!=0.17.1,!=0.17.2,>=0.14.0 (from jsonschema<4.18.0,>=3.0.1->opentrons_shared_data==8.2.0->opentrons-hardware@ git+https://github.com/Opentrons/opentrons@release#subdirectory=hardware)
  Obtaining dependency information for pyrsistent!=0.17.0,!=0.17.1,!=0.17.2,>=0.14.0 from https://files.pythonhosted.org/packages/15/ee/ff2ed52032ac1ce2e7ba19e79bd5b05d152ebfb77956cf08fcd6e8d760ea/pyrsistent-0.20.0-cp312-cp312-macosx_10_9_universal2.whl.metadata
  Downloading pyrsistent-0.20.0-cp312-cp312-macosx_10_9_universal2.whl.metadata (27 kB)
Downloading opentrons_shared_data-8.2.0-py2.py3-none-any.whl (723 kB)
   ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 723.6/723.6 kB 22.3 MB/s eta 0:00:00
Downloading pyserial-3.5-py2.py3-none-any.whl (90 kB)
   ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 90.6/90.6 kB 31.8 MB/s eta 0:00:00
Downloading jsonschema-4.17.3-py3-none-any.whl (90 kB)
   ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 90.4/90.4 kB 479.2 MB/s eta 0:00:00
Downloading pydantic-1.10.21-cp312-cp312-macosx_11_0_arm64.whl (2.5 MB)
   ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 2.5/2.5 MB 33.6 MB/s eta 0:00:00
Downloading typing_extensions-4.12.2-py3-none-any.whl (37 kB)
Downloading attrs-25.1.0-py3-none-any.whl (63 kB)
   ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 63.2/63.2 kB 461.6 MB/s eta 0:00:00
Downloading pyrsistent-0.20.0-cp312-cp312-macosx_10_9_universal2.whl (83 kB)
   ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 83.5/83.5 kB 8.6 MB/s eta 0:00:00
Building wheels for collected packages: opentrons-hardware
[... cut ...]
  Attempting uninstall: opentrons_shared_data
    Found existing installation: opentrons_shared_data 8.2.0
    Uninstalling opentrons_shared_data-8.2.0:
      Successfully uninstalled opentrons_shared_data-8.2.0
Successfully installed attrs-25.1.0 jsonschema-4.17.3 opentrons-hardware-8.2.0 opentrons_shared_data-8.2.0 pydantic-1.10.21 pyrsistent-0.20.0 pyserial-3.5 typing-extensions-4.12.2

[notice] A new release of pip is available: 23.2.1 -> 25.0
[notice] To update, run: pip install --upgrade pip
```

tobiasko, 2025-02-03:
```
(.venv) ~/PycharmProjects/opentronsflex git:[main]
opentrons_simulate protocols/enrich-iST_96x_HT/enrich-iST_beadPrep_enrich_lyse_8channel_EppendorfDeepWellPlate_96_500_uL.py -L ~/PycharmProjects/opentronsflex/customLabware/
Traceback (most recent call last):
  File "/Users/tobiaskockmann/PycharmProjects/opentronsflex/.venv/bin/opentrons_simulate", line 8, in <module>
    sys.exit(main())
             ^^^^^^
  File "/Users/tobiaskockmann/PycharmProjects/opentronsflex/.venv/lib/python3.12/site-packages/opentrons/simulate.py", line 1020, in main
    runlog, maybe_bundle = simulate(
                           ^^^^^^^^^
  File "/Users/tobiaskockmann/PycharmProjects/opentronsflex/.venv/lib/python3.12/site-packages/opentrons/simulate.py", line 555, in simulate
    with _make_hardware_simulator_cm(
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Library/Frameworks/Python.framework/Versions/3.12/lib/python3.12/contextlib.py", line 137, in __enter__
    return next(self.gen)
           ^^^^^^^^^^^^^^
  File "/Users/tobiaskockmann/PycharmProjects/opentronsflex/.venv/lib/python3.12/site-packages/opentrons/simulate.py", line 377, in _make_hardware_simulator_cm
    result = _make_hardware_simulator(override=None, robot_type=robot_type)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/tobiaskockmann/PycharmProjects/opentronsflex/.venv/lib/python3.12/site-packages/opentrons/simulate.py", line 350, in _make_hardware_simulator
    from opentrons.hardware_control.ot3api import OT3API
  File "/Users/tobiaskockmann/PycharmProjects/opentronsflex/.venv/lib/python3.12/site-packages/opentrons/hardware_control/ot3api.py", line 143, in <module>
    from .backends.flex_protocol import FlexBackend
  File "/Users/tobiaskockmann/PycharmProjects/opentronsflex/.venv/lib/python3.12/site-packages/opentrons/hardware_control/backends/flex_protocol.py", line 42, in <module>
    from opentrons_hardware.sensors.types import SensorDataType
  File "/Users/tobiaskockmann/PycharmProjects/opentronsflex/.venv/lib/python3.12/site-packages/opentrons_hardware/sensors/types.py", line 10, in <module>
    from opentrons_hardware.firmware_bindings.messages.fields import SensorTypeField
  File "/Users/tobiaskockmann/PycharmProjects/opentronsflex/.venv/lib/python3.12/site-packages/opentrons_hardware/firmware_bindings/messages/__init__.py", line 2, in <module>
    from .messages import MessageDefinition, get_definition
  File "/Users/tobiaskockmann/PycharmProjects/opentronsflex/.venv/lib/python3.12/site-packages/opentrons_hardware/firmware_bindings/messages/messages.py", line 7, in <module>
    from . import message_definitions as defs
  File "/Users/tobiaskockmann/PycharmProjects/opentronsflex/.venv/lib/python3.12/site-packages/opentrons_hardware/firmware_bindings/messages/message_definitions.py", line 8, in <module>
    from . import payloads
  File "/Users/tobiaskockmann/PycharmProjects/opentronsflex/.venv/lib/python3.12/site-packages/opentrons_hardware/firmware_bindings/messages/payloads.py", line 39, in <module>
    @dataclass(eq=False)
     ^^^^^^^^^^^^^^^^^^^
  File "/Library/Frameworks/Python.framework/Versions/3.12/lib/python3.12/dataclasses.py", line 1265, in wrap
    return _process_class(cls, init, repr, eq, order, unsafe_hash,
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Library/Frameworks/Python.framework/Versions/3.12/lib/python3.12/dataclasses.py", line 994, in _process_class
    cls_fields.append(_get_field(cls, name, type, kw_only))
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Library/Frameworks/Python.framework/Versions/3.12/lib/python3.12/dataclasses.py", line 852, in _get_field
    raise ValueError(f'mutable default {type(f.default)} for field '
ValueError: mutable default <class 'opentrons_hardware.firmware_bindings.utils.binary_serializable.UInt32Field'> for field message_index is not allowed: use default_factory
```

leoschwarz, 2025-02-04:
Turns out, this bug disappears with Python 3.10. It's reproducible from Python 3.11 onwards. While it's a completely different error, which might indicate a different cause, this is potentially related to the problem in #12006

tobiasko, 2025-02-05:
I can confirm this. @koji fix only works in combination with python 3.10.

```
(.venv1) ~/PycharmProjects/opentronsflex git:[main]
opentrons_simulate protocols/enrich-iST_96x_HT/enrich-iST_beadPrep_enrich_lyse_8channel_EppendorfDeepWellPlate_96_500_uL.py -L ~/PycharmProjects/opentronsflex/customLabware/
/Users/tobiaskockmann/.opentrons/robot_settings.json not found. Loading defaults
Belt calibration not found.
Unlatching labware on Heater-Shaker
Setting Temperature Module temperature to 4.0 °C (rounded off to nearest integer)
1. Beads preparation
Bead preparation cycle 1
Distributing 200.0 from A1 of Hamilton 5 Reservoir 48000 µL on slot D2 to A1 of Eppendorf 96 Well Plate 500 µL on slot B2
        Transferring 200.0 from A1 of Hamilton 5 Reservoir 48000 µL on slot D2 to A1 of Eppendorf 96 Well Plate 500 µL on slot B2
                Picking up tip from A1 of Opentrons Flex 96 Tip Rack 1000 µL on slot C3
                Aspirating 805.0 uL from A1 of Hamilton 5 Reservoir 48000 µL on slot D2 at 716.0 uL/sec
                Dispensing 200.0 uL into A1 of Eppendorf 96 Well Plate 500 µL on slot B2 at 716.0 uL/sec
                Dispensing 200.0 uL into A2 of Eppendorf 96 Well Plate 500 µL on slot B2 at 716.0 uL/sec
                Dispensing 200.0 uL into A3 of Eppendorf 96 Well Plate 500 µL on slot B2 at 716.0 uL/sec
                Dispensing 200.0 uL into A4 of Eppendorf 96 Well Plate 500 µL on slot B2 at 716.0 uL/sec
                Blowing out into Trash Bin on slot A3
                Aspirating 805.0 uL from A1 of Hamilton 5 Reservoir 48000 µL on slot D2 at 716.0 uL/sec
                Dispensing 200.0 uL into A5 of Eppendorf 96 Well Plate 500 µL on slot B2 at 716.0 uL/sec
                Dispensing 200.0 uL into A6 of Eppendorf 96 Well Plate 500 µL on slot B2 at 716.0 uL/sec
                Dispensing 200.0 uL into A7 of Eppendorf 96 Well Plate 500 µL on slot B2 at 716.0 uL/sec
                Dispensing 200.0 uL into A8 of Eppendorf 96 Well Plate 500 µL on slot B2 at 716.0 uL/sec
                Blowing out into Trash Bin on slot A3
                Aspirating 805.0 uL from A1 of Hamilton 5 Reservoir 48000 µL on slot D2 at 716.0 uL/sec
                Dispensing 200.0 uL into A9 of Eppendorf 96 Well Plate 500 µL on slot B2 at 716.0 uL/sec
                Dispensing 200.0 uL into A10 of Eppendorf 96 Well Plate 500 µL on slot B2 at 716.0 uL/sec
                Dispensing 200.0 uL into A11 of Eppendorf 96 Well Plate 500 µL on slot B2 at 716.0 uL/sec
                Dispensing 200.0 uL into A12 of Eppendorf 96 Well Plate 500 µL on slot B2 at 716.0 uL/sec
                Blowing out into Trash Bin on slot A3
                Dropping tip into Trash Bin on slot A3
Moving eppendorf_96_wellplate_500ul_5 to opentrons_universal_flat_adapter with gripper
Latching labware on Heater-Shaker
Setting Heater-Shaker to Shake at 1200 RPM and waiting until reached
Delaying for 1 minutes and 0.0 seconds
Deactivating Shaker
Unlatching labware on Heater-Shaker
Moving eppendorf_96_wellplate_500ul_5 to MagneticBlockContext at Magnetic Block GEN1 on C1 lw None with gripper
Delaying for 2 minutes and 0.0 seconds
Processing well: A1 of Eppendorf 96 Well Plate 500 µL on Magnetic Block GEN1 on slot C1in wash cycle: 1
Picking up tip from A1 of Opentrons Flex 96 Tip Rack 200 µL on slot A1
Aspirating 200.0 uL from A1 of Eppendorf 96 Well Plate 500 µL on Magnetic Block GEN1 on slot C1 at 71.60000000000001 uL/sec
Dispensing 200.0 uL into A1 of NEST 1 Well Reservoir 195 mL on slot C2 at 716.0 uL/sec
Blowing out at A1 of NEST 1 Well Reservoir 195 mL on slot C2
Returning tip
        Dropping tip into A1 of Opentrons Flex 96 Tip Rack 200 µL on slot A1
Processing well: A2 of Eppendorf 96 Well Plate 500 µL on Magnetic Block GEN1 on slot C1in wash cycle: 1
Picking up tip from A2 of Opentrons Flex 96 Tip Rack 200 µL on slot A1
Aspirating 200.0 uL from A2 of Eppendorf 96 Well Plate 500 µL on Magnetic Block GEN1 on slot C1 at 71.60000000000001 uL/sec
Dispensing 200.0 uL into A1 of NEST 1 Well Reservoir 195 mL on slot C2 at 716.0 uL/sec
Blowing out at A1 of NEST 1 Well Reservoir 195 mL on slot C2
Returning tip
        Dropping tip into A2 of Opentrons Flex 96 Tip Rack 200 µL on slot A1
Processing well: A3 of Eppendorf 96 Well Plate 500 µL on Magnetic Block GEN1 on slot C1in wash cycle: 1
Picking up tip from A3 of Opentrons Flex 96 Tip Rack 200 µL on slot A1
Aspirating 200.0 uL from A3 of Eppendorf 96 Well Plate 500 µL on Magnetic Block GEN1 on slot C1 at 71.60000000000001 uL/sec
Dispensing 200.0 uL into A1 of NEST 1 Well Reservoir 195 mL on slot C2 at 716.0 uL/sec
Blowing out at A1 of NEST 1 Well Reservoir 195 mL on slot C2
Returning tip
        Dropping tip into A3 of Opentrons Flex 96 Tip Rack 200 µL on slot A1
Processing well: A4 of Eppendorf 96 Well Plate 500 µL on Magnetic Block GEN1 on slot C1in wash cycle: 1
Picking up tip from A4 of Opentrons Flex 96 Tip Rack 200 µL on slot A1
Aspirating 200.0 uL from A4 of Eppendorf 96 Well Plate 500 µL on Magnetic Block GEN1 on slot C1 at 71.60000000000001 uL/sec
Dispensing 200.0 uL into A1 of NEST 1 Well Reservoir 195 mL on slot C2 at 716.0 uL/sec
Blowing out at A1 of NEST 1 Well Reservoir 195 mL on slot C2
Returning tip
        Dropping tip into A4 of Opentrons Flex 96 Tip Rack 200 µL on slot A1
Processing well: A5 of Eppendorf 96 Well Plate 500 µL on Magnetic Block GEN1 on slot C1in wash cycle: 1
[... 464 lines cut ...]
[... cut ...]
Aspirating 100.0 uL from A5 of Hamilton 5 Reservoir 48000 µL on slot D2 at 716.0 uL/sec
Dispensing 100.0 uL into A1 of Eppendorf 96 Well Plate 500 µL on Opentrons Universal Flat Heater-Shaker Adapter on Heater-Shaker Module GEN1 on slot D1 at 716.0 uL/sec
Dropping tip into Trash Bin on slot A3
Setting Heater-Shaker to Shake at 1200 RPM and waiting until reached
Delaying for 1 minutes and 0.0 seconds
Deactivating Shaker
Unlatching labware on Heater-Shaker
```

```
(.venv1) ~/PycharmProjects/opentronsflex git:[main]
python --version
Python 3.10.11
```

koji, 2025-02-06:
@tobiasko we haven't supported python 3.11 yet.