Opentrons/opentrons #17437: bug: OverPressure Errors

kostrouc, 2025-02-05:
### Overview

Is it possible to override the over-pressure errors on the Flex? I have labware that previously worked fine until the over-pressure feature was implemented. I would like a way to disable this error-checking feature if possible.

### Operating system

None

Jonpot, 2025-05-07:
Bump, this is needed.

chris-discopharma, 2025-07-30:
I would like to know more about the option to disable this - @osand317 , is this a GUI option or is it only accesible via the API? We are using stable channel v8.5.1.