Opentrons/opentrons #18182: feat: Glibc version on OT2

Neil-YL, 2025-04-25:
### Overview

We’re trying to run Prefect natively on the OT-2 for workflow orchestration. However, we ran into compatibility issues because the robot’s glibc version is 2.25, while Prefect‘s dependencies require 2.28+.

```
ldd --version
ldd (GNU libc) 2.25
$Copyright (C) 2017 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
$Written by Roland McGrath and Ulrich Drepper. 
```

### Implementation details

For more details I have opened an issue on[ package request](https://github.com/Opentrons/opentrons-python-packages/issues/11).

### Design

Would it be possible to include a more recent version of glibc in future releases? Alternatively, is there a safe way for us to upgrade it on our side?

sfoster1, 2025-05-21:
There's not really a safe way for you to upgrade the glibc version for the robot as a whole. What you could do is get an appropriately compiled glibc version, drop it in locally, and then run prefect with an `LD_PRELOAD` that points to it.

Compiling it appropriately will be a bit of a pain but you might be able to get it from a recent raspbian release that supports the rpi3b; glibc is nice and stable and so you could put it on a pi and copy it from `/lib` (or just flash it to an sd card and browse the sd card from a linux machine).

Neil-YL, 2025-06-13:
Copying from email thread as Opentrons support suggested:

-------

I finally got a chance to have a x86_64 Windows PC (running with WSL) to try the build tools in https://github.com/Opentrons/opentrons-python-packages/tree/main/tools. I successfully built the Docker container using the provided Dockerfile, and then attempted to build a new package, prefect, under the packages/ directory.

I created a build.py file with all necessary dependencies for the patched pendulum and the target Prefect version. However, when I run the build command, I consistently receive the following warning:

```
<frozen runpy>:128: RuntimeWarning: 'builder.package_build.orchestrate' found in sys.modules after import of package 'builder.package_build', but prior to execution of 'builder.package_build.orchestrate'; this may result in unpredictable behaviour.
```

And unfortunately, the build process does not proceed beyond this point.

I’ve tried several workarounds I found online, but none of them resolved the issue. I’m relatively new to Docker, though I’ve previously managed to cross-compile a Prefect wheel on my M2 Mac using a Docker environment that mimicked the OT-2 system. I assumed the build tools approach would be similar, but I seem to be stuck.

It seems like the issue might be related to how builder/package_build/orchestrate.py is being called, and I’m not confident enough with the internals to modify it safely. If you have any suggestions or a more detailed guidance on how to use the build tools, I’d really appreciate your help.

sfoster1, 2025-06-13:
Responded over here: https://github.com/Opentrons/opentrons-python-packages/issues/11 I'm not terribly optimistic, unfortunately. Those are a lot of dependencies, and glibc versioning can be tough.